sharandac/My-TTGO-Watch #2: Firmware wont compile

32u4, 2020-07-18:
Trying to compile with arduino IDE.
This error occurs:

Arduino: 1.8.13 (Linux), Board: "TTGO T-Watch, Enabled, Default (2 x 6.5 MB app, 3.6 MB SPIFFS), 2000000, None"

sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z5setupv+0x50): undefined reference to `motor_setup()'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z5setupv+0x54): undefined reference to `display_setup(TTGOClass*)'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z5setupv+0x58): undefined reference to `screenshot_setup()'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z5setupv+0x5c): undefined reference to `splash_screen_stage_one(TTGOClass*)'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z5setupv+0x60): undefined reference to `splash_screen_stage_update(char const*, int)'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z5setupv+0x64): undefined reference to `powermgm_setup(TTGOClass*)'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z5setupv+0x68): undefined reference to `splash_screen_stage_finish(TTGOClass*)'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z5setupv+0x6c): undefined reference to `gui_setup()'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z5setupv+0x70): undefined reference to `display_get_brightness()'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z4loopv+0x0): undefined reference to `gui_loop(TTGOClass*)'
sketch/my-ttgo-watch.ino.cpp.o:(.literal._Z4loopv+0x4): undefined reference to `powermgm_loop(TTGOClass*)'
sketch/my-ttgo-watch.ino.cpp.o: In function `setup()':
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:24: undefined reference to `motor_setup()'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:31: undefined reference to `display_setup(TTGOClass*)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:33: undefined reference to `screenshot_setup()'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:35: undefined reference to `splash_screen_stage_one(TTGOClass*)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:36: undefined reference to `splash_screen_stage_update(char const*, int)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:38: undefined reference to `splash_screen_stage_update(char const*, int)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:40: undefined reference to `splash_screen_stage_update(char const*, int)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:44: undefined reference to `splash_screen_stage_update(char const*, int)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:46: undefined reference to `splash_screen_stage_update(char const*, int)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:47: undefined reference to `powermgm_setup(TTGOClass*)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:48: undefined reference to `splash_screen_stage_update(char const*, int)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:49: undefined reference to `splash_screen_stage_finish(TTGOClass*)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:51: undefined reference to `gui_setup()'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:54: undefined reference to `display_get_brightness()'
sketch/my-ttgo-watch.ino.cpp.o: In function `loop()':
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:63: undefined reference to `gui_loop(TTGOClass*)'
/home/user1/Arduino/Watch/my-ttgo-watch/my-ttgo-watch.ino:64: undefined reference to `powermgm_loop(TTGOClass*)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board TTGO T-Watch.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

sharandac, 2020-07-18:
Please use vscode with platformio as described in the readme.md. This is not a project for the arduino IDE and support for the arduino IDE is not planned for the future.

closed 2020-07-18 by sharandac

32u4, 2020-07-19:
OK thanks.
Got it to compile. 
Great job on the improvements!