sharandac/My-TTGO-Watch #3: can´t compile

ondragu, 2020-07-18:
Tried to build and upload.

Error is :
collect2.exe: error: ld returned 1 exit status

What can I do ?

sharandac, 2020-07-18:
Do you use vscode with platformio? If so, which version?

ondragu, 2020-07-18:
Version: 1.47.2 (user setup)
Commit: 17299e413d5590b14ab0340ea477cdd86ff13daf
Date: 2020-07-15T18:22:06.216Z
Electron: 7.3.2
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.18363

sharandac, 2020-07-18:
Thanks. This looks good so far. Would you have a complete build log with all messages?

ondragu, 2020-07-18:
> Executing task in folder My-TTGO-Watch-master: C:\Users\libid\.platformio\penv\Scripts\platformio.exe run <

Processing ttgo-t-watch (platform: espressif32; board: ttgo-t-watch; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-watch.html
PLATFORM: Espressif 32 1.12.4 > TTGO T-Watch
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TTGO TWatch Library> 1.2.0
|   |-- <Wire> 1.0.1
|   |-- <Ticker> 1.1
|   |-- <SPI> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncTCP> 1.1.1
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <SD(esp32)> 1.0.5
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <Ticker> 1.1
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <Update> 1.0
Building in release mode
Linking .pio\build\ttgo-t-watch\firmware.elf
.pio\build\ttgo-t-watch\src\my-ttgo-watch\font\Ubuntu_16px.c.o:(.data.Ubuntu_16px+0x0): multiple definition of `Ubuntu_16px'
.pio\build\ttgo-t-watch\src\font\Ubuntu_16px.c.o:(.data.Ubuntu_16px+0x0): first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\font\Ubuntu_72px.c.o:(.data.Ubuntu_72px+0x0): multiple definition of `Ubuntu_72px'
.pio\build\ttgo-t-watch\src\font\Ubuntu_72px.c.o:(.data.Ubuntu_72px+0x0): first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\gui\gui.cpp.o: In function `gui_setup()':
C:\Users\libid\Downloads\My-TTGO-Watch-master\My-TTGO-Watch-master/src\my-ttgo-watch\gui/gui.cpp:20: multiple definition of `gui_setup()'
.pio\build\ttgo-t-watch\src\gui\gui.cpp.o:C:\Users\libid\Downloads\My-TTGO-Watch-master\My-TTGO-Watch-master/src\gui/gui.cpp:20: first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\gui\gui.cpp.o: In function `gui_loop(TTGOClass*)':
gui.cpp:(.text._Z8gui_loopP9TTGOClass+0x0): multiple definition of `gui_loop(TTGOClass*)'
.pio\build\ttgo-t-watch\src\gui\gui.cpp.o:gui.cpp:(.text._Z8gui_loopP9TTGOClass+0x0): first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\gui\keyboard.cpp.o: In function `keyboard_setup()':
C:\Users\libid\Downloads\My-TTGO-Watch-master\My-TTGO-Watch-master/src\my-ttgo-watch\gui/keyboard.cpp:8: multiple definition of `keyboard_setup()'
.pio\build\ttgo-t-watch\src\gui\keyboard.cpp.o:C:\Users\libid\Downloads\My-TTGO-Watch-master\My-TTGO-Watch-master/src\gui/keyboard.cpp:8: first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\gui\keyboard.cpp.o: In function `keyboard_hide()':
keyboard.cpp:(.text._Z13keyboard_hidev+0x0): multiple definition of `keyboard_hide()'
.pio\build\ttgo-t-watch\src\gui\keyboard.cpp.o:keyboard.cpp:(.text._Z13keyboard_hidev+0x0): first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\gui\keyboard.cpp.o: In function `keyboard_show()':
keyboard.cpp:(.text._Z13keyboard_showv+0x0): multiple definition of `keyboard_show()'
.pio\build\ttgo-t-watch\src\gui\keyboard.cpp.o:keyboard.cpp:(.text._Z13keyboard_showv+0x0): first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\gui\keyboard.cpp.o: In function `keyboard_set_textarea(_lv_obj_t*)':
keyboard.cpp:(.text._Z21keyboard_set_textareaP9_lv_obj_t+0x0): multiple definition of `keyboard_set_textarea(_lv_obj_t*)'
.pio\build\ttgo-t-watch\src\gui\keyboard.cpp.o:keyboard.cpp:(.text._Z21keyboard_set_textareaP9_lv_obj_t+0x0): first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\gui\mainbar\app_tile\app_tile.cpp.o: In function `app_tile_setup(_lv_obj_t*, lv_style_t*, short, short)':
C:\Users\libid\Downloads\My-TTGO-Watch-master\My-TTGO-Watch-master/src\my-ttgo-watch\gui\mainbar\app_tile/app_tile.cpp:12: multiple definition of `app_tile_setup(_lv_obj_t*, lv_style_t*, short, short)'
.pio\build\ttgo-t-watch\src\gui\mainbar\app_tile\app_tile.cpp.o:C:\Users\libid\Downloads\My-TTGO-Watch-master\My-TTGO-Watch-master/src\gui\mainbar\app_tile/app_tile.cpp:12: first defined here
[... 328 lines cut ...]
.pio\build\ttgo-t-watch\src\my-ttgo-watch\images\unlock_16px.c.o:(.rodata.unlock_16px_map+0x0): multiple definition of `unlock_16px_map'
.pio\build\ttgo-t-watch\src\images\unlock_16px.c.o:(.rodata.unlock_16px_map+0x0): first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\images\wifi_64px.c.o:(.rodata.wifi_64px+0x0): multiple definition of `wifi_64px'
.pio\build\ttgo-t-watch\src\images\wifi_64px.c.o:(.rodata.wifi_64px+0x0): first defined here
.pio\build\ttgo-t-watch\src\my-ttgo-watch\images\wifi_64px.c.o:(.rodata.wifi_64px_map+0x0): multiple definition of `wifi_64px_map'
.pio\build\ttgo-t-watch\src\images\wifi_64px.c.o:(.rodata.wifi_64px_map+0x0): first defined here
[... cut ...]
C:\Users\libid\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:17: undefined reference to `loop()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\ttgo-t-watch\firmware.elf] Error 1
================================================================================================== [FAILED] Took 47.25 seconds ==================================================================================================
The terminal process "C:\Users\libid\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

ondragu, 2020-07-19:
I have forked it now directly from VS through the github addon and now it works !!

Many thanks for your help and your project.

Greetings
Christopher from Aachen

sharandac, 2020-07-19:
Thanks for the feedback. I've already tried to verify the error, but without success. I would close the ticket.

closed 2020-07-19 by sharandac