sharandac/My-TTGO-Watch #4: recompiling: foot_16px undefined reference

aircable, 2020-07-20:
compiling the latest source leads to:
Linking .pio/build/ttgo-t-watch/firmware.elf
.pio/build/ttgo-t-watch/src/gui/statusbar.cpp.o:(.literal._Z15statusbar_setupv+0x68): undefined reference to `foot_16px'

sharandac, 2020-07-20:
Thanks for your bug report. Unfortunately I cannot reproduce the bug. Could you please check if the file foot_16px.c can be found under src/images/ ?

aircable, 2020-07-21:
Thanks for your reply. I think it's a dependency problem. 

The steps were: clone repo, then just: pio run -t upload

Here is my graph:
Dependency Graph
```
|-- <TTGO TWatch Library> 1.2.0
|   |-- <Wire> 1.0.1
|   |-- <Ticker> 1.1
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncTCP> 1.1.1
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <SD(esp32)> 1.0.5
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <Ticker> 1.1
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <Update> 1.0
```

The files are all there, cloned from your repo. For some reason the files are not getting compiled.
I just reinstalled my platformIO and it still does the same.

I have the full compile log and this foot_16px.c does not get compiled.

[compile_log.txt](https://github.com/sharandac/My-TTGO-Watch/files/4950566/compile_log.txt)
 
Also there are those warnings: Setup45_TTGO_T_Watch.h:26:0: warning: "SPI_FREQUENCY" redefined

Where do you define dependencies? Only those images files get compiled, nothing else in images:

Compiling .pio/build/ttgo-t-watch/src/images/battery_icon_64px.c.o
Compiling .pio/build/ttgo-t-watch/src/images/move_32px.c.o
Compiling .pio/build/ttgo-t-watch/src/images/move_64px.c.o

sharandac, 2020-07-21:
I have just installed vscode and platformio on another computer. There it built without any further problems.
I think it is not a dependency problem. The images have no dependencies and stand for themselves, they are pure data.

Here is my logfile:

[complie.txt](https://github.com/sharandac/My-TTGO-Watch/files/4950651/complie.txt)

I think this is a problem with vscode or platformio. Do you have the possibility to try it on another computer?

The dependencies are defined in the platformio.ini . Here you can also disable the warning "SPI_FREQUENCY" redefined. Simply comment out the following line in the platformio.ini:

`-DSPI_FREQUENCY=80000000L`

aircable, 2020-07-21:
Is it possible that this warning causes some issues?
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name

It happens pretty much at every compile file.

I will try on a Linux machine next. Thanks for helping.

sharandac, 2020-07-21:
The warnings is not so critical. It only warns you that definitions contain illegal characters, for example. For example:

`#define _FOOBAR.H`

generates a warning.  In C this would be ok, but in C++ it is a bit more pedantic and not very welcome. Correct would be:

`#define _FOOBAR_H`

passes the compiler without warning. It is also popular not to have an empty line at the end of a file. This generates a similar error message.

However, these warnings are generated by the libraries used.

Good luck with testing on a Linux machine. I would be glad about a feedback.

closed 2020-07-23 by sharandac

aircable, 2020-07-23:
Sorry it took so long. A linux machine was able to correctly compile the repo. 
I would consider reopening the case and change the headline to MAC OSX not able to compile, even though it might be a PlatformIO problem.