sharandac/My-TTGO-Watch #10: TTGO Watch 2020 hard reboots on connecting to Wifi (core 0 panic)

robjohnc, 2020-07-23:
T-watch 2020 hard reboots every time it connects to Wifi. 
It appears to successfully sync RTC, as the time changes and the RTC sync message is output to Serial.
Serial Monitor Displays the following:

```
syncToRtc: 120 7 24 - 12 54 11 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000142d  PS      : 0x00060a30  A0      : 0x800d5f9d  A1      : 0x3ffe1a80
A2      : 0x3ffc1a6d  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x3ffc1a6d  A11     : 0x3f450164  A12     : 0x00000000  A13     : 0x3ffe16d0
A14     : 0x3ffe18d0  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000142d  LEND    : 0x4000143a  LCOUNT  : 0xffffffff

Backtrace: 0x4000142d:0x3ffe1a80 0x400d5f9a:0x3ffe1a90 0x400d4bcb:0x3ffe2ad0 0x4008d59d:0x3ffe2af0

Rebooting...

```

and then the usual bootup sequence...
I don't think the RTC sync is causing the issue, as it still occurs when this is disabled.

sharandac, 2020-07-23:
Opps ... Which version of My-TTGO-Watch did you use? The last one?

robjohnc, 2020-07-23:
Yes, downloaded this morning.

sharandac, 2020-07-23:
Okay. If you have the possibility to use a newer version now, that would be great. The development is progressing quite fast right now. In the settings you will find the update icon and there also the version number. 202072307 is up to date.

EverlastEngineering, 2020-07-25:
Hmm, I'm getting the same problem but in the new weather code a little after the RTC work.

 ```
syncToRtc: 120 7 22 - 10 58 21 
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (weather sync Ta) 
Core 0 register dump:
PC      : 0x40087eec  PS      : 0x00060f36  A0      : 0x3ffdd680  A1      : 0x3ffdd5c0  
A2      : 0x00000000  A3      : 0x00060f23  A4      : 0x00060f20  A5      : 0x0000001c  
A6      : 0x00060d20  A7      : 0x00000000  A8      : 0x80089c52  A9      : 0x3ffdd750  
A10     : 0x0000001c  A11     : 0x40089c0c  A12     : 0x00000084  A13     : 0x3fff09ac  
A14     : 0x0000000a  A15     : 0x00000008  SAR     : 0x0000001f  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000
 ```

32u4, 2020-07-25:
Reboots occur with latest firmware too.
Compiled and uploaded without any modifications. 
As soon as wifi connects the device reboots.

supermenteur, 2020-07-25:
Same for me :(

reality52, 2020-07-26:
Hi guys! Im new here.
Had the same problem and couple of others on a latest firmware (072408)
With weather autoupdates turned off it works almost like a charm with no reboots after connection to WiFi and NTP Syncing. 
But when i set up again weather autoupdates in its settings it is works normal after that and updates RTC and weather after waking up with no reboot.
(i have an issue with wrong time and date from time to time, so it must always  be connected to wifi for showing correct time, but i think is another story)

robjohnc, 2020-07-26:
Ah, it seems that it's the weather doing it for me also, but additionally when Wifi does connect, it then hard reboots 10-20 seconds after you click the update button on the firmware page.

reality52, 2020-07-26:
> 
> 
> Ah, it seems that it's the weather doing it for me also, but additionally when Wifi does connect, it then hard reboots 10-20 seconds after you click the update button on the firmware page.

Same, COM monitors shows it before reset:
`                               [E][WiFiClient.cpp:288] setSocketOption(): 1006 :                                 9
`

sharandac, 2020-07-27:
Please test the latest version ( 2020072703 ) and give me some feedback. fixed:

- improved task synchronization
- core voltage raised to 3.3V on wakeup and 3.0V on standby
- weather synchronization error handling

robjohnc, 2020-07-27:
Seems to be resolved, at least for me.

tbendiksen, 2020-07-27:
> Seems to be resolved, at least for me.

For me also.

sharandac, 2020-07-27:
thanks for you feedback! rtc is also fixed now.

closed 2020-07-27 by sharandac