sharandac/My-TTGO-Watch #6: How to use guide?

EverlastEngineering, 2020-07-23:
This compiled and uploaded just fine, but I don't know how to use it. I can swipe down from the top and reveal "Scan.." or "Activate" with a WiFi logo, and that's it. 

How do I scroll to the other settings, etc?

sharandac, 2020-07-23:
With a simple left, right, up or down swipe you can switsch between the 4 main screen (main tile, app tile, note tile and setup tile). it is like tinder :)

EverlastEngineering, 2020-07-23:
Interesting, so somethings off for me. Left / Right swipes do not register. Can you point to where i can drop in some debug console msgs for swipes so I can dig it to find the problem?

EverlastEngineering, 2020-07-23:
Just played more and *just* got one sideways swipe to work... but painful. If you can point to the swiping code I'll debug from there as to the senstivity, detection, etc.

EverlastEngineering, 2020-07-23:
(Could be my greasy fat fingers too lol)

AlexGoodyear, 2020-07-23:
Hi, I had the same issues but I can get better reliability if ...
 - from the watch screen you only swipe left in the lower third of the screen (below the time). Swiping down from here is also recognised.
- The "note" and "app" screens seem ok to swipe anywhere.
- The setup screen is more difficult, you have to try and make sure you start a swipe on a bare bit of background otherwise the swipe is ignored or you select a button.

sharandac, 2020-07-23:
Yes, it's not for big fingers :) a problem with swiping is that blocks every element on the screen in a very impractical way, so that unused space is needed. Is more of an LVGL problem, or a fat finger problem. lol

closed 2020-07-24 by sharandac

EverlastEngineering, 2020-07-25:
Having coded a couple of touch interfaces from scratch early on in the web days in js, I can see what they've done wrong with insufficient "when to ignore" code that needs to pass through details of a touch to the underlying items, or use a master overlay which can detect the location and pass that to logic that can fire an event for the correctly placed button, etc. I'll dig in and see if I can lend a help to those defects in LVGL from that end.

krokoko, 2020-08-01:
I'm facing the same issue, able to get access to the weather and note tile, but not the others.