sharandac/My-TTGO-Watch #7: Keyboard is impossible to use

aircable, 2020-07-23:
This is more of a usability issue then a "bug" in the code. The keyboard implementation is really impossible to use, even with a stylized pen since the touch sensor isn't accurate enough. 
It took me like over 15 minutes to type in the SSID.

Please consider making the keyboard class like this: (see [](https://github.com/AlexGoodyear/agoodWatch/blob/master/agoodWatch/gui.cpp))

```
const char *Keyboard::btnm_mapplus[10][23] = {
    {
        "a", "b", "c",   "\n",
        "d", "e", "f",   "\n",
        "g", "h", "i",   "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    },
    {
        "j", "k", "l", "\n",
        "n", "m", "o",  "\n",
        "p", "q", "r",  "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    },
    {
        "s", "t", "u",   "\n",
        "v", "w", "x", "\n",
        "y", "z", " ", "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    },
    {
        "A", "B", "C",  "\n",
        "D", "E", "F",   "\n",
        "G", "H", "I",  "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    },
    {
        "J", "K", "L", "\n",
        "N", "M", "O",  "\n",
        "P", "Q", "R", "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    },
    {
        "S", "T", "U",   "\n",
        "V", "W", "X",   "\n",
        "Y", "Z", " ", "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    },
    {
        "0", "1", "2",  "\n",
        "3", "4", "5",  "\n",
        "6", "7", "8",  "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    },
    {
        "9", "+", "-",  "\n",
        "/", "*", "=",  "\n",
        "!", "?", "#",  "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    },
    {
        "<", ">", "@",  "\n",
        "%", "$", "(",  "\n",
        ")", "{", "}",  "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    },
    {
        "[", "]", ";",  "\n",
        "\"", "'", ".", "\n",
        ",", ":",  " ", "\n",
        LV_SYMBOL_OK, "Del", "Exit", LV_SYMBOL_RIGHT, ""
    }
};
```

sharandac, 2020-07-23:
me too! But it's not a priority right now. I'll make a note of it. Sounds like a first feature request :-)

duenf33, 2020-07-23:
keyboard is perfect for me. I am able to type correctly and fast anything, however, the issue im having is with the watch not displaying the battery charge level and it keeps updating randomly here and there and resets. I was checking the code and its missing the include file or folder.
"/Users/username/Documents/PlatformIO/Projects/My-TTGO-Watch/include".

closed 2020-07-24 by sharandac