sharandac/My-TTGO-Watch #11: Some wishes

JoanMCD, 2020-07-26:
Great that someone start writing a SW for this piece of HW. I love the first version, and the WiFi PW Keyboard is unbelievable well working !

Some wishes:
- set the focus on a stable, working and long working watch function (the time is often lost - RTC used?)
  (if the base unctions are given it make sense to wear it and from time to time check for updates.)
- if on power, enable WiFi, open Web server, sync time and use a longer display-on time
- show more information like in battery life info  (over all the IP address, gyro and sensors, license and project,... )
- Web server, have a index.htm, may use bootstrab, do the most settings via web (it's more easy) 
- have a concept to sync with Smartphone (notifications) 
- change positive things to green color (bootscreen, sliders bullets, switch backgrounds)
- Update - switch WiFi (if on power) automatically on (update not working yet ?)
- (Arduino IDE would be more common)
- App to Apps :)

BR Joe

reality52, 2020-07-27:
`the time is often lost` i had that problem, its because it was a bad connection on RTC small battery, i put a small piece of paper on it and it works good now) started using this watch for 1 day now without NTP syncing, time is ticking good)

JoanMCD, 2020-07-27:
On the 2020 watch the battery is soldered

reality52, 2020-07-27:
> On the 2020 watch the battery is soldered

In my case negative contact was unsoldered from the board and battery was flapping.

mailliard, 2020-07-28:
because its difficult to type the wifi password where can I fix it inside teh code itself ?
thank you

JoanMCD, 2020-07-29:
Much better :) 
To be more relaxed on IP/Webserver tasks, can you something add to keep the WiFi on under line power.

JoanMCD, 2020-07-29:
[image]

sharandac, 2020-07-30:
@mailliard I change the keyboard size and input method, so i hope it works better.
@JoanMCD Nice!

JoanMCD, 2020-07-30:
@sharandac Yes, it's placed in the SPIFSS, and (later) interact by websockets (true variant from Links2004) to fill the values (running with ___ MHz). I will send you some pics from a other project to your PM.

I need something to keep the watch 'online' if powered. I.E. a #define or later a switch on the webpage.
 
One more request(wish): It's possible to change the format of the config files to a readable / editable format later. (low priority).

JoanMCD, 2020-08-06:
It’s in your interest?

sharandac, 2020-08-06:
Theoretically, yes. But just a little early for this project for my feelings. And right now I have no time to program more than the basic framework. There is only sporadic time for the hobby.
The timeout can currently be set to infinite in the display menu.

JoanMCD, 2020-08-06:
OK, fine - I'm sure that you are more active as me. :-)  I need less programming support from you. Lets start with small things - what's like ?:
- Upnp presentation in Windows, 40lines of code + SSDP library
- WPS function (Wi-Fi Protected Setup), 100 lines of code.

Later:
- Webpage structure based on AsynWebServer (here the question is how to update without overwrite the config )
- REST/Get/ URL call  as UserApp
- the 'Smart' part of the watch ...
- your / other wishes ?

Display Timeout; Great!! :+1:

bwagstaff, 2020-08-06:
FWIW, I am putting that same web server demo web page in my own fork. I will upload it to Github tonight after work. I am also putting together a few games as well. This project felt like a great one to use as a base. 

WPS is supported by the library but has no interface, it would be a good first project for someone willing to get hands dirty.

sharandac, 2020-08-06:
WPS is done in 2020080605. Only tested with my Fritzbox 7490.

bwagstaff, 2020-08-06:
Oh, in case people didn't know it already, there is some basic file manipulation already in the server under {ipaddress}/edit

sharandac, 2020-08-06:
Yeah! It is a nice feature and comes with AsyncWebserver :+1:

bwagstaff, 2020-08-06:
PR #21 created from my branch for a friendlier web home page.  

Currently the page has security through obscurity, you must know the actual URLs on the device to trigger them. Putting out a friendly index page makes that slightly more concerning when on wifi.  At some point a bit of security, like a setup page with a user/password pair and encryption and more will be wanted, but for now that's what we can offer.

Comments in the pull request, but if you don't want to wipe out your existing flash image, navigate to `{ipaddress}\edit` on your device with the current firmware, then upload `index.htm` and `favicon.ico` manually.  The info data is just stuff I've found useful while developing. And of course, go ahead and replace the image with whatever, that's just the one from the demos.

Elekronixx, 2020-08-07:
To clearify - the list 5 field above is what I can contribute to the project - not what I request from you :) 
But anyway - it's welcome. 

**WPS**: Super and fast integration !
**Editor**: Great tool, thts a reason why I ask for clear text config files :) One Info, in the original implementation the editor loads files from external sources (google?Don't remember). I had modified it and place the files into the SPIFSS.
**WebGui**: Big afford with less amount of time = good work ! 👍 
@bwagstaff Here you touch the problem I also saw. The config files. If it will keep being only 2 files the editor upload is a good way. OK -it's not visible on my screenshot but my webpage is not static, it uses JS + CSS libraries for Bootstrap, Gauges, input checking, life update, communication with back end, ... So many files in the SPIFSS. Best updated in sync with the FW. But the configfiles ... What do you think ? 
a) update all the files by an mechanism that the online ESP downloads it and write it to the FS
b) move the config to EEPROM (FS possible ?)
c) find a way to backup the config (load it into the Structs/Objects/Memory), perform the update (,do not restart), write it back
d) save it to PC (export/import)
c) store it online (!?!)
...
x) kick it :-(

Joe

JoanMCD, 2020-08-07:
To clearify - the list 5 field above is what I can contribute to the project - not what I request from you :) 
But anyway - it's welcome. 

**WPS**: Super and fast integration !
**Editor**: Great tool, thts a reason why I ask for clear text config files :) One Info, in the original implementation the editor loads files from external sources (google?Don't remember). I had modified it and place the files into the SPIFSS.
**WebGui**: Big afford with less amount of time = good work ! 👍 
@bwagstaff Here you touch the problem I also saw. The config files. If it will keep being only 2 files the editor upload is a good way. OK -it's not visible on my screenshot but my webpage is not static, it uses JS + CSS libraries for Bootstrap, Gauges, input checking, life update, communication with back end, ... So many files in the SPIFSS. Best updated in sync with the FW. But the configfiles ... What do you think ? 
a) update all the files by an mechanism that the online ESP downloads it and write it to the FS
b) move the config to EEPROM (FS possible ?)
c) find a way to backup the config (load it into the Structs/Objects/Memory), perform the update (,do not restart), write it back
d) save it to PC (export/import)
c) store it online (!?!)
...
x) kick it :-(

Joe

JoanMCD, 2020-08-07:
the Upnp feature:
[Upnp presentation / SSDP](https://github.com/sharandac/My-TTGO-Watch/issues/23)
 
Joe

bwagstaff, 2020-08-07:
The web editor is the one from the web server demo, it appears unmodified. That makes it very convenient, but nothing customized.   

As I've needed to touch config files I've tried to write data migrations. Most didn't include a version field and blindly required a file size. While that works for a product that is static, it doesn't work to well for dynamic projects where save data is occasionally modified, needing to be migrated.

SPIFFS is somewhat difficult regarding updating individual files.  When flashing and updating tools tend to treat it as an all-or-nothing data partition, rather than patching individual files.

JoanMCD, 2020-08-07:
I agree, also the management and cleanup is heavy

~~a) update all the files by an mechanism that the online ESP downloads it and write it to the FS~~
b) move the config to EEPROM (FS possible ?)
c) find a way to backup the config (load it into the Structs/Objects/Memory), perform the update (,do not restart), write it back
d) save it to PC (export/import)
c) store it online (!?!)
...
~~x) kick it :-(~~

JoanMCD, 2020-08-07:
Editor: it's not what I mean - also unmodified it downloads from external server: 

[image]

It's not only bad, is saves flash and increases speed - but it's good to know.

bwagstaff, 2020-08-07:
The files exist in the source tree, already compressed and minimized, and you have the option to copy them from there.  The thing is, the device only has 3,604,480 bytes for the SPIFFS partition.  Personally I'd rather not use 200KB for those files.  

That's about 5% of the available storage space to serve something that can easily served from elsewhere.

bwagstaff, 2020-08-07:
Oh, as for moving configuration files from SPIFFS to NVS.  While in some ways that is the purpose of NVS, it is much harder to modify the NVS partition and it isn't organized as files, just a blob of data.  That would be a bigger pain for expanding it out unless there is coordination for layout and use of the data, as any change in ANY app that stores configuration value would break the NVS format layout.  

We've done that type of thing in projects at work, but even with (paid) professionals maintaining it occasionally it would still break badly, requiring everyone to wipe all their values.  At least with config files on SPIFFS you can work with them individually, and (barring accidentally reformatting your SPIFFS partition) you only lose the individual config file you corrupted.

JoanMCD, 2020-08-08:
So - whats you offer ?
The list gets shorter. And at the end we should have more than: not possible.
Other ideas ?

JoanMCD, 2020-08-08:
One point to NVS. It's the right point to store the network credentials. In the config file, accessible by editor not only the watch is vulnerable - the hole Network security is gone.

closed 2020-08-18 by sharandac