sharandac/My-TTGO-Watch #13: Strange power readings

reality52, 2020-07-27:
Hi! I have some strange power readings. Voltage seems to be good, i measured battery with multimeter. Sometimes it shows "?" in statusbar instead of 0%.I had this bug on older firmware too. Sometimes, when i connect watch to PC "0%" resets to  random number but never shows 100% no matter how long i will charge it. VBat tho can be around 4.2V. And i can go with 0% all day...

[image]

Anyone have this bug?
Sorry for photo, i cant take screenshots yet :(

sharandac, 2020-07-27:
this is not a real bug. the charge controller knows how much it has to charge. this is an attempt to determine the exact charge of the battery with the help of the current measurement. a calibration is probably necessary. for this purpose the device must be completely discharged once until it switches itself off.

reality52, 2020-07-27:
> this is not a real bug. the charge controller knows how much it has to charge. this is an attempt to determine the exact charge of the battery with the help of the current measurement. a calibration [...]

Thanks! I will do it as fast as i can!

closed 2020-07-27 by sharandac

reality52, 2020-07-30:
I know, that is closex, but I calibrated the battery ☺️i think its because in the code capacity is 300, but i have "380"... So i need to change capacity values in pwrmngmnt i guess... 
Which battery do u have guys? [image]
[image]

tbendiksen, 2020-07-30:
For what it's worth: I have the same battery model and capacity. 

I have not played much with this, but after a full cycle it seems to operate as expected and I have seen no issues with it. It charges to close to (and crucially stays below) 100%. 
The fact that the battery has a capacity of 380mAh and the software shows 300 I wouldn't think too much about as what the battery can deliver is not the same as what the hardware can use. This is actually more about what voltage a battery can deliver once above or below a certain amount the battery will be seen as "full" or empty, and we hope that the watch has been able to extract at least 300mAh from it in between "full" and "empty". 
I haven't looked at the code for this, but I assume that there is charge circuitry that takes care of charging the battery until it reaches a threshold (usually 4.2V as mentioned on the back of the battery). I suspect that your battery charge circuitry is out of whack and is confused.
Doing a few more cycles should normally clear things up - but this is based on my gut feeling based on previous projects and not on this HW or code in this repo (and I am no maintainer of this code, just a potential future contributor with a keen interest).