sharandac/My-TTGO-Watch #15: Using with T-Watch 2020

Sk3yNet, 2020-07-29:
First, my many thanks for the work you have put into this firmware. I've done a successful upload to my T-Watch 2020, but I found the line in config.h "To use T-Watch2020, please uncomment this line". I've tried uncommenting just that line or the whole block of code but then get errors when building:

.pio/libdeps/ttgo-t-watch/TTGO TWatch Library/src/LilyGoWatch.h:177:2: error: #error "Please define Watch model"
 #error "Please define Watch model"

Just wondering if that is expected behavior. Any guidance would be appreciated!

tbendiksen, 2020-07-30:
This is expected behaviour. In C header files comments start with "//" and defines start with # 
I have not been involved in this, but I believe this to be a leftover from the LilyGo code that this is based on (see https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/examples/LVGL/SimpleWatch/config.h as an example).

Saying that this should be cleaned up for clarity, but functionally this is correct and you should leave it as it was. This code only works for the TTGO T-Watch 2020 (unlike the LilyGo code which can also work on the very different 2019 model).

Sk3yNet, 2020-07-31:
Great, thank you!

closed 2020-07-31 by Sk3yNet