sharandac/My-TTGO-Watch #16: make the watch less energy consuming

d3th3n, 2020-07-30:
I found several solutions to make the watch much less energy consuming:
    on powermgm.cpp change the function as is :

void powermgm_loop(TTGOClass *ttgo)
{

    // event-tripper pmu-button or pmu-battery state change
    if (powermgm_get_event(POWERMGM_PMU_BUTTON | POWERMGM_PMU_BATTERY | POWERMGM_BMA_WAKEUP | POWERMGM_SILENCE_WAKEUP_REQUEST | POWERMGM_STANDBY_REQUEST))
    {

        // if we have an request when we are in silence mode, emulate an wakeup from standby
        if (powermgm_get_event(POWERMGM_SILENCE_WAKEUP) && powermgm_get_event(POWERMGM_PMU_BUTTON | POWERMGM_PMU_BATTERY | POWERMGM_BMA_WAKEUP))
        {
            powermgm_set_event(POWERMGM_STANDBY);
            powermgm_clear_event(POWERMGM_SILENCE_WAKEUP);
        }

        if (powermgm_get_event(POWERMGM_STANDBY))
        {
            powermgm_clear_event(POWERMGM_STANDBY);
            ttgo->power->setDCDC3Voltage(3300);
            // normal wake up from standby
            if (powermgm_get_event(POWERMGM_PMU_BUTTON | POWERMGM_PMU_BATTERY | POWERMGM_BMA_WAKEUP))
            {
                log_i("wakeup");
                display_go_wakeup(ttgo);
                motor_vibe(1);
            }
            // silence wakeup request from standby
            else if (powermgm_get_event(POWERMGM_SILENCE_WAKEUP_REQUEST))
            {
                log_i("silence wakeup");
                display_go_silence_wakeup(ttgo);
                powermgm_set_event(POWERMGM_SILENCE_WAKEUP);
            }
            timesyncToSystem();
            ttgo->startLvglTick();
            mainbar_jump_to_maintile(LV_ANIM_OFF);
            lv_disp_trig_activity(NULL);
            if (bma_get_config(BMA_STEPCOUNTER))
                ttgo->bma->enableStepCountInterrupt(true);
            //wifictl_on();                                         // for stopping the wifi on each time you switch on your watch
            ttgo->power->clearTimerStatus();
            ttgo->power->offTimer();

            // doing loop                                          // Adding loop management ONLY if the watch is switched on
            pmu_loop(ttgo);
            bma_loop(ttgo);
            display_loop(ttgo);
        }
        else
        {
            log_i("go to standby");
            display_go_sleep(ttgo);
            timesyncToRTC();
            if (powermgm_get_event(POWERMGM_WIFI_ACTIVE)) wifictl_off();
            while (powermgm_get_event(POWERMGM_WIFI_ACTIVE | POWERMGM_WIFI_CONNECTED | POWERMGM_WIFI_OFF_REQUEST | POWERMGM_WIFI_ON_REQUEST | POWERMGM_WIFI_SCAN))
            {
                yield();
            }
            ttgo->stopLvglTick();
            if (bma_get_config(BMA_STEPCOUNTER))
                ttgo->bma->enableStepCountInterrupt(false);
            powermgm_set_event(POWERMGM_STANDBY);
            powermgm_clear_event(POWERMGM_SILENCE_WAKEUP);
            ttgo->power->setDCDC3Voltage(3000);
            ttgo->power->clearTimerStatus();
            ttgo->power->setTimer(60);
            setCpuFrequencyMhz(10);
            gpio_wakeup_enable((gpio_num_t)AXP202_INT, GPIO_INTR_LOW_LEVEL);
            gpio_wakeup_enable ((gpio_num_t)BMA423_INT1, GPIO_INTR_HIGH_LEVEL);
            esp_sleep_enable_gpio_wakeup();
            esp_light_sleep_start();
        }
        // clear event
        powermgm_clear_event(POWERMGM_PMU_BUTTON | POWERMGM_PMU_BATTERY | POWERMGM_BMA_WAKEUP | POWERMGM_STANDBY_REQUEST | POWERMGM_SILENCE_WAKEUP_REQUEST);
    }
    else {                                                       // and if the watch is switched ON (not off !)
        pmu_loop(ttgo);
        bma_loop(ttgo);
        display_loop(ttgo);
    }
}

we really have a real gain on energy consumption ... One more thing ... in the TTGO T-Watch 2020 the battery is 380mA ... not 300mA

sharandac, 2020-07-30:
Thanks for your suggestion. But I don't think you quite understand the code at this point. This is event-driven code, like a state-machine. In standby mode, after
```
setCpuFrequencyMhz(10);
gpio_wakeup_enable((gpio_num_t)AXP202_INT, GPIO_INTR_LOW_LEVEL);
gpio_wakeup_enable ((gpio_num_t)BMA423_INT1, GPIO_INTR_HIGH_LEVEL);
esp_sleep_enable_gpio_wakeup();
```
is executed, the CPU goes into sleep mode and no further code is executed. Here the watch consumes about 2mA in standby. I think that is ok. Only an Interrupt from bma or pmu wakes up the CPU.
The only point where I see a suggestion is that every time I turn on the watch, the wifi is turned on.

The 300mAh are caused by my own measurements and is only an information. The charge controller never charges more than 300mAh into the battery (see battery setup) and is limited by the final charge voltage of 4.2V.

d3th3n, 2020-07-30:
I agree with you but (without a measuring device) I can see a real difference in consumption. in principle, operation should "go into pause" but the battery barely lasts 6 hours but with the modification ... it really lasts longer (test to come)

Best regards

sharandac, 2020-07-30:
Oh... I'm a little surprised. With "normal" use, my battery lasts about two days. Without use it takes about 3 days until the battery is empty, that includes waking up every hour with wifi for 15 seconds.

d3th3n, 2020-07-30:
this remains true during the night (when I do not use the watch) the discharge of the battery is very low ... on the other hand, during the day the drainage of the battery is very important (hence the modification) but, can -be, that the problem is elsewhere (in one of the two "short-circuited" functions?). All this to tell you that I appreciate your project

Best regards

closed 2020-08-02 by sharandac