sharandac/My-TTGO-Watch #17: Big battery overcharge !

d3th3n, 2020-08-01:
Hello, after several charging / discharging of the battery ... I find myself with a battery charged to more than 140% (412.2mAh) while the manufacturer maximum is 300 to 380 mAh.

tbendiksen, 2020-08-01:
First of all this would be an issue with the hardware and as far as I know (as a user and not a developer of this project) there is no control over charging done here. I am not certain but I guess there is a hardware controller that deals with the charging circuitry. On my watch it seems to get awfully confused if I remove the battery. 

But separately from that this is not exactly how lithium batteries work.

Generally speaking the mAh measurement is not a limit or a guarantee, but rather something the manufacturer will print as a guideline. Think of it as the fuel tank in a car, but flexible in volume (more like a balloon). The charging circuitry will deal with upper and lower limits on the voltage (could loosely be compared to the fuel pressure in a car). Below a certain value and the circuitry will shut off and assume that the battery is dead. Higher than a set value and the battery is deemed to be full.
How much energy it can deliver between full and empty is judged as the capacity and this is measured in mili Ampere hours (mAh). The manufacturer states that the battery should be able to deliver at least 380mAh (the 300mAh in software I cannot speak to). This statement from the manufacturer is based not so much on their promises and beliefs but rather chemistry.

What is a little bit crucial is that there are a LOT of variables involved (especially sensitive to temperature) and some times you can either input more or less than the specified energy into the battery or you can extract more or less. This often happens, but over time this will even out regardless,but also as batteries get older they have less peaks like this.

So the TL;DR version is, this is probably normal, but it could be that the charging circuitry is not ideal or optimal. But this would likely be a hardware issue rather than an issue with this piece of software.

bwagstaff, 2020-08-02:
"while the manufacturer maximum is 300 to 380 mAh ... " 

Because that's not a thing being measured.  The chip actually measures voltage.  While the nominal voltage is 3.7V, it can reach up to around 4.2V on a completely full charge, and slowly drops down until the chip cuts out at around 3.2V. Be aware that hitting both the maximum possible charge and a complete battery drain are bad for this type of battery, shortening the useful life. 

Everything else is an estimate based on the measured voltage and the estimated power draw.

closed 2020-08-03 by sharandac

mikey60, 2020-08-03:
There is definitely a difference between this firmware and other versions like the original [LilyGo](https://github.com/Xinyuan-LilyGO/TTGO-T-Watch) version and the [agoodWatch](https://github.com/AlexGoodyear/agoodWatch) version.  I am able to get a few days of operation with the agoodWatch version after charging to 100%.  When I install this version, the watch only indicates 0-10% battery when the other two versions indicate 90-100%??