sharandac/My-TTGO-Watch #18: Return icon and vibration feedback

ondragu, 2020-08-03:
The return function, this means back from setups to pages is difficult to use and requires often 3 to 5 times tapping on it.
The vibration feedback should be switchable. 

I´m looking forward to new features and I must say a great "Thank you"!!!!
At the moment I´m learning to get in C++ programming and have admit how difficult it is .

Greetings from Aachen

Christopher

sharandac, 2020-08-03:
thanks for your feedback! the operation for the touch is a bit tricky, a bit inaccurate. the haptic feedback is currently independent from icons actually are pressed. a little confused. the edges of the touchscreen are rather imprecise. it takes some practice to hit the icon.

greetings from Berlin

Dirk

sharandac, 2020-08-03:
Can you test the latest version (2020080313, update available via setup) and give me some feedback?

ondragu, 2020-08-03:
Wow, great !
"Geht wie Schmitz´Katze"  ;-))

sharandac, 2020-08-03:
Four lines of code can do that!

ondragu, 2020-08-03:
Hope I understand them soon :-)

sharandac, 2020-08-03:
it wasn't magic. Scale up the x axis a little and you're done.

```
    float temp_x = ( x - ( LV_HOR_RES_MAX / 2 ) ) * 1.15;
    float temp_y = ( y - ( LV_VER_RES_MAX / 2 ) ) * 1.0;
    x = temp_x + ( LV_HOR_RES_MAX / 2 );
    y = temp_y + ( LV_VER_RES_MAX / 2 );
```

closed 2020-08-03 by sharandac

KrustyDerClown, 2022-02-11:
Hallo von Berlin nach Berlin, 
I installed latest self compiled firmware and I have immense trouble to hit the buttons horizontaly. Mostly I had to touch more right at the left side of the screen and on the right side of the screen vice versa. I found this "workaround" that seems to fit to this shift of the touch result I experience at the moment. I  completely comment it out. and now for me it a lot better. 

Did the hardware change 2020v3 or am I a untypical toucher?

DevFlorian, 2022-04-02:
Hi KrustyDerClown,
i'm also having Trouble With hitting Buttons in V3. What exactly did you commented out?

KrustyDerClown, 2022-04-04:
Hey Florian,

see https://github.com/sharandac/My-TTGO-Watch/issues/18#issuecomment-668176105

"Uncomment" is not correct ;)

replace this unnatural 1.15 by 1.0, this works for me

lewisxhe, 2022-05-06:
I changed the x-axis scaling ratio from 1.15 to 1.0 or 0.8 in V2, V3, it just felt right, 1.15 is not suitable for everyone