sharandac/My-TTGO-Watch #23: PR: Upnp presentation / SSDP

JoanMCD, 2020-08-07:
Hi, here is some small code to to the watch be viable in the Network in Windows. A double click opens the webpage. In the context menu you get additional information (it not exact matches the title i.e. serial number -> SW version, modelnumber -> hostname). 

This allows you to run the watch on DHCP and/or find it easily in you network neighborhood.
!! The watch must on Wi-Fi !!

@sharandac feel free to change the content.

**Pictures**
[image]

[image]

**How to implement**
in PlatformIO, install ESP32SSDP library from me-no-dev

webserver.h: 
`#define UPNPPORT 80`

webserver.c: 

top:
`#include <ESP32SSDP.h> // ESP32 upnp`

somewhere:

```
  //Upnp presentation website
  //#todo - use global information:

#define DEV_NAME "My-Watch" 
#define DEV_INFO "Watch based on ESP32 from Espressif Systems"

  asyncserver.on("/description.xml", HTTP_GET, [](AsyncWebServerRequest *request) {
    byte mac[6];
    WiFi.macAddress(mac);
    char tmp[12 + 1];
    snprintf(tmp, sizeof(tmp), "%02X%02X%02X%02X%02X%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
    String MacStrPart = String(tmp).substring(6);

    String xmltext = String("<?xml version=\"1.0\"?>\n") +
            "<root xmlns=\"urn:schemas-upnp-org:device-1-0\">\n"
            "<specVersion>\n"
            "\t<major>1</major>\n"
            "\t<minor>0</minor>\n"
            "</specVersion>\n"
            "<URLBase>http://" + WiFi.localIP().toString() + "/</URLBase>\n" 
            "<device>\n"
            "\t<deviceType>upnp:rootdevice</deviceType>\n"

            /*this is the icon name in Windows*/
            /*"\t<friendlyName>" + WiFi.getHostname() + "</friendlyName>\n"*/ 
            "\t<friendlyName>" + DEV_NAME + " " + MacStrPart + "</friendlyName>\n" /*because the hostename is 'Espressif' */

            "\t<presentationURL>/</presentationURL>\n"
            "\t<manufacturer>" + "Dirk Broßwick (sharandac)" + "</manufacturer>\n"
            "\t<manufacturerURL>https://github.com/sharandac/My-TTGO-Watch</manufacturerURL>\n"
            "\t<modelName>" +  DEV_INFO + "</modelName>\n"

            "\t<modelNumber>" + WiFi.getHostname() + "</modelNumber>\n"
            "\t<modelURL>" +
            "/" + "</modelURL>\n"

            "\t<serialNumber>Build: " + __FIRMWARE__ + "</serialNumber>\n"
            //The last six bytes of the UUID are the hardware address of the first Ethernet adapter in the system the UUID was generated on.
            "\t<UDN>uuid:38323636-4558-4DDA-9188-CDA0E6" + MacStrPart + "</UDN>\n"
            "</device>\n"
            "</root>\r\n"
            "\r\n";

    request->send(200, "text/xml", xmltext);
  });
```

before the 'asyncserver.begin();':

```
  //Upnp / SSDP presentation - Multicast  - link to description.xml
  SSDP.setSchemaURL("description.xml");
  SSDP.setHTTPPort(UPNPPORT);
  SSDP.setURL("/");
  SSDP.setDeviceType("upnp:rootdevice");
  SSDP.begin();
```
Hope you find it useful !
 Joe

sharandac, 2020-08-07:
has been merged. thanks for the advance work.

closed 2020-08-07 by sharandac

datacute, 2020-08-16:
A minor improvement is to create `tmp` 6 characters long instead of 12, and remove the substring call.
```
    char tmp[6 + 1];
    snprintf(tmp, sizeof(tmp), "%02X%02X%02X", mac[0], mac[1], mac[2]);
    String MacStrPart = String(tmp);
```

JoanMCD, 2020-08-16:
Hi Datacute - looks smarter and does the same in this case.
. 
_The reason why it happens here is, that I grab it from one of my more generic helper functions and use it here. 
Normal the return String length is dynamic._ 
```
String getMACPart_Str(uint8_t from)
{
  byte mac[6];
  WiFi.macAddress(mac);
  char tmp[12 + 1];
  snprintf(tmp, sizeof(tmp), "%02X%02X%02X%02X%02X%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
  return String(tmp).substring(from);
}
```

Here the Upnp is more autonomous and you optimization make sense. :+1: 
 Thx for reading and the feedback !
Joe

datacute, 2020-08-17:
My optimisation _nearly_ does the same thing - and would if the parameters were fixed to be mac[3], mac[4], mac[5]