sharandac/My-TTGO-Watch #25: Swipes very hard

cranphin, 2020-08-07:
Hi!

With the current version, swipes seem to be quite poor, they only trigger with trying a lot.
The keyboard is pretty good, so accuracy seems fine.

It's especially annoying trying to get from the main screen down, since it triggers the weather configuration all the time instead :)
Is there any trick? :) Fast/slow, where to start/end? Do you recognize this, or could it be a hardware issue?

I know the firmware is new, just letting you know :)

cranphin, 2020-08-07:
Ah, I think I got it, the weather 'icon' on the home screen is eating the swipe gestures, if I swipe to the left or right of it, it seems fine :)

sharandac, 2020-08-07:
It is a problem with LVGL. icon, labels, button and so on eating swipe gestures.

closed 2020-08-07 by sharandac