sharandac/My-TTGO-Watch #26: PR: web pages (index, info, network, frameset, ... )

JoanMCD, 2020-08-07:
HI, based on the work from bwagstaff I:

- Enhance info page
- Create network information page
- Move index.htm to c-code (for now) to provide it faster to all users by the well working update 
 (and add /network link)  - better design will follow :)

@bwagstaff I hope it's OK 

Also, it's a nice collection of ESP32 functions :-)

Install:
- add following code to webserver.c (  replace asyncserver.on("/info" .... ){}   )

```
  asyncserver.on("/index.htm", HTTP_GET, [](AsyncWebServerRequest *request) {
    String html = (String) "<!DOCTYPE html>"
      "<html>"
      "<frameset cols=\"300, *\">"
      "<frame src=\"/nav.htm\" name=\"nav\">"
      "<frame name=\"cont\">"
      "</frameset>"
      "</html>";
    request->send(200, "text/html", html);
  });

asyncserver.on("/nav.htm", HTTP_GET, [](AsyncWebServerRequest *request) {
    String html = (String) "<!DOCTYPE html>"
      "<html><head>"
      "<meta http-equiv='Content-type' content='text/html; charset=utf-8'>"
      "<title>Web Interface</title>"
      "</head><body>"
      "<h1>TTGo Watch Web Server</h1>"
      "<p>This is your device, program it as you see fit."
      "<p>Here are some URLs the device already supports, which you might find helpful:"
      "<ul>"
      "<li><a target=\"cont\" href=\"/info\">/info</a> - Display information about the device"
      "<li><a target=\"cont\" href=\"/network\">/network</a> - Display network information"
      "<li><a target=\"cont\" href=\"/shot\">/shot</a> - Capture a screen shot"
      "<li><a target=\"cont\" href=\"/screen.565\">/screen.565</a> - Retrieve the image in RGB565 format"
      "<li><a target=\"cont\" href=\"/edit\">/edit</a> - View, edit, upload, and delete files"
      "</ul>"
      "<p><div style=\"color:red;\">Caution:</div> Use these with care:"
      "<ul><li><a href=\"/reset\">/reset</a> Reboot the device"
      "<li><a href=\"/update\">/update</a> Transmit a firmware update through POST request"
      "</body></html>";
    request->send(200, "text/html", html);
  });

  asyncserver.on("/info", HTTP_GET, [](AsyncWebServerRequest *request) {
    FlashMode_t mode = ESP.getFlashChipMode();
    int SketchFull = ESP.getSketchSize() + ESP.getFreeSketchSpace();

    String html = (String) "<html><head><meta charset=\"utf-8\"></head><body><h3>Information</h3>" +
                  "<b><u>Memory</u></b><br>" +
                  "<b>Heap size: </b>" + MemFormat(ESP.getHeapSize()) + "<br>" +
                  "<b>Heap free: </b>" + MemFormat(ESP.getFreeHeap()) + "<br>" +
                  "<b>Heap free min: </b>" + MemFormat(ESP.getMinFreeHeap()) + "<br>" +
                  "<b>Heap size: </b>" + MemFormat(ESP.getHeapSize()) + "<br>" +
                  "<b>Psram size: </b>" + MemFormat(ESP.getPsramSize()) + "<br>" +
                  "<b>Psram free: </b>" + MemFormat(ESP.getFreePsram()) + "<br>" +

                  "<br><b><u>System</u></b><br>" +
                  "\t<b>Battery voltage: </b>" + TTGOClass::getWatch()->power->getBattVoltage() / 1000 + " Volts" + "<br>" +

                  "\t<b>Uptime: </b>" + time_str( millis() / 1000 ) + "<br>" +
                  "<br><b><u>Chip</u></b>" +
                  "<br><b>SdkVersion: </b>" + String(ESP.getSdkVersion()) + "<br>" +
                  "<b>CpuFreq: </b>" + String(ESP.getCpuFreqMHz()) + " MHz<br>" +
                  
                  "<br><b><u>Flash</u></b><br>" +
                  "<b>FlashChipSpeed: </b>" + String(ESP.getFlashChipSpeed() / 1000000) + " MHz<br>" +
                  "<b>Flash mode: </b>" + String( mode == FM_QIO ? "QIO" : mode == FM_QOUT ? "QOUT" : mode == FM_DIO ? "DIO" : mode == FM_DOUT ? "DOUT" : "UNKNOWN") + "</b><br>" +
                  "<b>Flash sector size: </b>" + String( SPI_FLASH_SEC_SIZE) + "<br>" +
                  "<b>FlashChipMode: </b>" + ESP.getFlashChipMode() + "<br>" +
                  "<b>FlashChipSize (SDK): </b>" + MemFormat(ESP.getFlashChipSize()) + "<br>" +
                  
                  "<br><b><u>Firmware</u></b><br>" +
                  "<b>SketchSpace free: </b>" + MemFormat(ESP.getFreeSketchSpace()) + " (" + (ESP.getFreeSketchSpace() / (SketchFull / 100)) + "%)<br>" +
                  "<b>BuildTime: </b>" +  __DATE__ + " " + __TIME__  + "<br>" +
                  "<b>Version: </b>" + __FIRMWARE__ + "<br>" +
                  "<b>GCC-Version: </b>" + __VERSION__ + "<br>" +
                  "<b>SketchMD5: </b>" + String(ESP.getSketchMD5()) + "<br>" +
                  
                  "<br><b><u>Filesystem</u></b><br>" +
                  "<b>Total size: </b>" + MemFormat(SPIFFS.totalBytes()) + "<br>" +
                  "<b>Used size: </b>" + MemFormat(SPIFFS.usedBytes()) + "<br>" +

                  "<br>";
    request->send(200, "text/html", html);
  });

  asyncserver.on("/network", HTTP_GET, [](AsyncWebServerRequest *request) {
    FlashMode_t mode = ESP.getFlashChipMode();
    int SketchFull = ESP.getSketchSize() + ESP.getFreeSketchSpace();
    char PHYmode[] = {'B', 'G', 'N'};
    String html = (String) "<html><head><meta charset=\"utf-8\"></head><body><h3>Network</h3>" +
                  "<b>IP Addr: </b>" + WiFi.localIP().toString() + "<br>" +
                  "<b>MAC: </b>" + WiFi.macAddress() + "<br>" +
                  "<b>SNMask: </b>" + WiFi.subnetMask().toString() + "<br>" +
                  "<b>GW IP: </b>" + WiFi.gatewayIP().toString() + "<br>" +
                  "<b>DNS 1: </b>" + WiFi.dnsIP(0).toString() + "<br>" +
                  "<b>DNS 2: </b>" + WiFi.dnsIP(1).toString() + "<br>" +
                  "<b>RSSI: </b>" + String(WiFi.RSSI()) + "dB<br>" +
                  "<b>Hostname: </b>" + WiFi.getHostname() + "<br>" +
                  "<b>SSID: </b>" + WiFi.SSID() + "<br>" +
                  "<br>Upnp Info: <a target=\"_blank\" href='/description.xml'>description.xml</a>" + "<br>" +
                  "</body></head></html>";
    request->send(200, "text/html", html);
  });
```

- remove index.htm (not the favicon) from data if exists

regards, Joe

bwagstaff, 2020-08-07:
My first thought is that most of those don't add anything useful, several aren't dynamic values. Most are meh to me. 

SDK version is known and constant, CPU frequency is adjusted based on tasks, the hardware does not change so flash chip layout is pointless. The sketch space is not really critical as the actual sizes used in flash and were listed already. Showing the IP address in the response is not useful as it was required for the web request in the first place.

IP address is visible in the network connection drop down, and MAC address is part of the network setup, although they may be useful on the page. 

I am only really against the last one, which I can easily disable on my branch. Never blindly serve the entire file system without security policies in place. Just because it is an open source IoT device does not make it wise for a blanket opening of all data. Even the edit page provides filtering.

bwagstaff, 2020-08-08:
Additionally, making the response HTML doesn't add features, but does increase executable size, consuming valuable memory.

JoanMCD, 2020-08-08:
We can discuss about ever single line. I simply print out whats known. The IP address is not 'known' (needed) if you double click the Upnp Icon, and maybe later it will be possible to configure it by the web. And it makes in my eyes no sense to print the network configuration and then skip the IP address because it is also visible in the drop down ! More - I have more watches, exactly IP in the web gives me the link to the right watch. So, kick what you don't like and save size and memory :)

Actually its an development device - security policies ?!. Where the 'serve the entire file system' is more open as before ?

Be constructive, thx.

sharandac, 2020-08-08:
I think we need a way in between. Enabling the web server as an opt-in would be one option. This way the device is more secure by default and it would be a deliberate action to turn it on.

@JoanMCD: where are `memformat` and `time_str` defined?

I will have a look at it on Monday, then more comments from me. Now weekend and off into nature.

bwagstaff, 2020-08-08:
I was pondering the same thing. Maybe another panel in setup, enable/disable file serving, with entry fields for http authentication. Espressif has encryption hardware support and libraries for HTTPS, generating a self-signed server. It quickly gets into a big mess of functionality consuming development time and device space, and is likely something I'd just leave off.

closed 2020-08-10 by sharandac

JoanMCD, 2020-08-10:
these are helper functions.  - I miss to add it. 

```
String time_str(long secs)
{
  long days = 0;
  long hours = 0;
  long mins = 0;
  String t_str = "";

  mins = secs / 60; //convert seconds to minutes
  hours = mins / 60; //convert minutes to hours
  days = hours / 24; //convert hours to days
  secs = secs - (mins * 60); //subtract the coverted seconds to minutes in order to display 59 secs max
  mins = mins - (hours * 60); //subtract the coverted minutes to hours in order to display 59 minutes max
  hours = hours - (days * 24); //subtract the coverted hours to days in order to display 23 hours max

  //Display results
  if (days > 0) // days will displayed only if value is greater than zero
  {
    t_str += (String)days + "d ";
  }

  if (hours <= 9)t_str += "0";
  t_str += (String)hours + "h ";
  if (mins <= 9)t_str += "0";
  t_str += (String)mins + "m ";
  if (secs <= 9)t_str += "0";
  t_str += (String)secs + "s";
  return t_str;
}
```

```
//format bytes
String formatBytes(size_t bytes) {
  if (bytes < 1024) {
    return String(bytes) + F("B");
  } else if (bytes < (1024 * 1024)) {
    return String(bytes / 1024.0) + F("KB");
  } else if (bytes < (1024 * 1024 * 1024)) {
    return String(bytes / 1024.0 / 1024.0) + F("MB");
  } else {
    return String(bytes / 1024.0 / 1024.0 / 1024.0) + F("GB");
  }
}

Thx, for adding !
```