sharandac/My-TTGO-Watch #28: HTTPS - Free memory to low

chrismcna, 2020-08-08:
Hi,

im trying to compete a https get request but i get the error below.
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():190]: (-32512) SSL - Memory allocation failed
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -32512

Im using 1.2 with the code below, ive tried increasing the "Stack size in words" but either i still get the same error or the task fails. The https request works fine in a stand alone program 

TaskHandle_t _test_Task;
void test_Task( void * pvParameters ) {
    Serial.println("enter_example_app_event_cb");
            Serial.println(ESP.getFreeHeap());
            
            HTTPClient http;
            http.useHTTP10( true );
            http.begin("https://*********"); //Specify the URL
            http.addHeader("Authorization","*******");
            int httpCode = http.GET();  
        
            if (httpCode != 200 ) { 
                static const char * btns[] ={"Close", ""};

                lv_obj_t * msgBox = lv_msgbox_create(lv_scr_act(), NULL);
                lv_msgbox_set_text(msgBox, "Failed");
                lv_msgbox_add_btns(msgBox, btns);
                lv_obj_set_width(msgBox, 200);
                lv_obj_align(msgBox, NULL, LV_ALIGN_CENTER, 0, 0);
            }
        

            http.end(); //Free the resources

            vTaskDelete( NULL );
}

static void enter_example_app_event_cb( lv_obj_t * obj, lv_event_t event ) {
    switch( event ) {
        case( LV_EVENT_CLICKED ):     

            xTaskCreate(    test_Task,      /* Function to implement the task */
                "example Task",    /* Name of the task */
                5000,                            /* Stack size in words */
                NULL,                            /* Task input parameter */
                1,                               /* Priority of the task */
                &_test_Task );  /* Task handle. */ 

            
            break;
    }    
}

chrismcna, 2020-08-08:
Sorry I've been stupid with the stack size, as increasing reduces the heap for the https connection.
To free up heap memory i had to comment out asyncwebserver_setup()

now https get is successfully completing

chrismcna, 2020-08-09:
with the LILYGO-TGO-T-Watch-2020 having 8MB of PSRAM i wonder if we can make use if it. 
As currently your software doesnt have much free memory for more apps, as i want to make some.

Currently i have added simple Amazon Alexa support for trigger Alexa routines from the watch, but want to make it as an "app" for your watch firmware

chrismcna, 2020-08-09:
Ive been reading PSRAM and we should be able to get the heap to be internal RAM + external RAM.
That would give us over 4MB of ram. I will continue looking another day

[image]

sharandac, 2020-08-09:
Or you can set a define in

`config.h`

like

`#define TWATCH_USE_PSRAM_ALLOC_LVGL`

this moves the framebuffer from LVGL into PSRAM and save a lot of memory. in the meantime the performance loss is hardly visible.

bwagstaff, 2020-08-09:
For reference, that shifts about 48KB into the 4MB PSRAM block, bringing available heap space up to around 105KB, more or less.

sharandac, 2020-08-09:
After all. I'm just looking if the PSRAM can be integrated into the hrap. But with the Arduino framework it does not work out of the box. I'm thinking about leaving the Arduino framework completely and using only the ESP-IDF. But this requires some rewrite and dirty hands.

bwagstaff, 2020-08-10:
I don't know how much it would buy you. It would limit adoption, as PIO is very standard. 

Either way, the device also only accesses 4MB of the 8 as a poo as isl, all external memory beyond the first 4 needs to be paged in. Sadly the chip does not support a built in page fault for VM to implement a lower level auto paging system.

All ESP32 chips with >4MB have the issue. It means those pages can be problematic without some kind of guard or mutex, or a policy of some sort. Probably as apps, active apps could use it exclusively as they go as a mutually agreed policy.

closed 2020-08-14 by sharandac