sharandac/My-TTGO-Watch #29: Random Crashes

chrismcna, 2020-08-09:
Hi, 
Not sure if your aware but i get random crashes

syncToRtc: 120 8 9 - 10 8 15 
[E][weather_fetch.cpp:101] weather_fetch_forecast(): HTTPClient error 401
[E][weather_fetch.cpp:56] weather_fetch_today(): HTTPClient error 401
syncToRtc: 120 8 9 - 10 18 26 
[E][weather_fetch.cpp:56] weather_fetch_today(): HTTPClient error 401
[E][weather_fetch.cpp:101] weather_fetch_forecast(): HTTPClient error 401
syncToRtc: 120 8 9 - 10 19 42 
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e4163  PS      : 0x00060f30  A0      : 0x800e49f4  A1      : 0x3ffb1b80  
A2      : 0x00000000  A3      : 0x00000010  A4      : 0x3f45b71c  A5      : 0x3ffb1d60  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000010  A13     : 0x3f45b71c  
A14     : 0x00000005  A15     : 0x00000004  SAR     : 0x00000001  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400ec960  LEND    : 0x400ec988  LCOUNT  : 0x0000000e  

Backtrace: 0x400e4163:0x3ffb1b80 0x400e49f1:0x3ffb1cd0 0x400f3c78:0x3ffb1d30 0x400e2e36:0x3ffb1df0 0x400e2ec7:0x3ffb1e40 0x400e2f1c:0x3ffb1e90 0x400e2f84:0x3ffb1eb0 0x400e344e:0x3ffb1ee0 0x400ecaeb:0x3ffb1f30 0x400ecc05:0x3ffb1f50 0x400d4b4d:0x3ffb1f70 0x400db021:0x3ffb1f90 0x4011386d:0x3ffb1fb0 0x4008d721:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
starting t-watch V1, version: 2020080702
Total heap: 242136
Free heap: 160272
Total PSRAM: 4194236
Free PSRAM: 4079036
[E][weather_fetch.cpp:56] weather_fetch_today(): HTTPClient error 401
[E][weather_fetch.cpp:101] weather_fetch_forecast(): HTTPClient error 401
syncToRtc: 120 8 9 - 10 20 13

chrismcna, 2020-08-09:
I've been attempting to look into this, but been struggling to decode the back trace.
But I've finally found a setting for Platformio to help. 
add "monitor_filters= default, esp32_exception_decoder" to platform.ini

ill leave the monitor connected and see if i can get anything useful

chrismcna, 2020-08-09:
Core 1 register dump:
PC      : 0x401d6f0e  PS      : 0x00060a30  A0      : 0x800e948b  A1      : 0x3ffb1b90
A2      : 0x00000000  A3      : 0x00000082  A4      : 0x00000000  A5      : 0x000000fb
A6      : 0x3ffb95fc  A7      : 0x00000000  A8      : 0xffffffff  A9      : 0x00000020  
A10     : 0x3ffcee70  A11     : 0x00000010  A12     : 0x3ffcee68  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x3ffd4ab4  SAR     : 0x00000001  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x400e72c4  LEND    : 0x400e72cf  LCOUNT  : 0x0000000e

Backtrace: 0x401d6f0e:0x3ffb1b90 0x400e9488:0x3ffb1bb0 0x400e2ad4:0x3ffb1cb0 0x400f1c41:0x3ffb1d30 0x400e31a6:0x3ffb1d50 0x400e3237:0x3ffb1da0 0x400e3237:0x3ffb1df0 0x400e3237:0x3ffb1e40 0x400e328c:0x3ffb1e90 0x400e32f4:0x3ffb1eb0 0x400e37be:0x3ffb1ee0 0x400ece5b:0x3ffb1f30 0x400ecf75:0x3ffb1f50 0x400d4c19:0x3ffb1f70 0x400db311:0x3ffb1f90 0x4011460d:0x3ffb1fb0 0x4008d74d:0x3ffb1fd0
  #0  0x401d6f0e:0x3ffb1b90 in _lv_memset_ff at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_misc/lv_mem.c:779
  #1  0x400e9488:0x3ffb1bb0 in draw_border at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_draw/lv_draw_rect.c:482
      (inlined by) lv_draw_rect at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_draw/lv_draw_rect.c:99
  #2  0x400e2ad4:0x3ffb1cb0 in lv_obj_design at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_obj.c:2916
  #3  0x400f1c41:0x3ffb1d30 in lv_btn_design at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_widgets/lv_btn.c:176
  #4  0x400e31a6:0x3ffb1d50 in lv_refr_obj at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_refr.c:167
  #5  0x400e3237:0x3ffb1da0 in lv_refr_obj at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_refr.c:167
  #6  0x400e3237:0x3ffb1df0 in lv_refr_obj at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_refr.c:167
  #7  0x400e3237:0x3ffb1e40 in lv_refr_obj at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_refr.c:167
  #8  0x400e328c:0x3ffb1e90 in lv_refr_obj_and_children at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_refr.c:167
  #9  0x400e32f4:0x3ffb1eb0 in lv_refr_area_part at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_refr.c:167
  #10 0x400e37be:0x3ffb1ee0 in lv_refr_area at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_refr.c:435
      (inlined by) lv_refr_areas at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_refr.c:357
      (inlined by) _lv_disp_refr_task at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_core/lv_refr.c:198
  #11 0x400ece5b:0x3ffb1f30 in lv_task_exec at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_misc/lv_task.c:331
  #12 0x400ecf75:0x3ffb1f50 in lv_task_handler at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_misc/lv_task.c:331
  #13 0x400d4c19:0x3ffb1f70 in gui_loop(TTGOClass*) at src\gui/gui.cpp:94
  #14 0x400db311:0x3ffb1f90 in loop() at C:/Users/Chris/Documents/GitHub/My-TTGO-Watch/src/my-ttgo-watch.ino:94
  #15 0x4011460d:0x3ffb1fb0 in loopTask(void*) at C:\Users\Chris\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:19
  #16 0x4008d74d:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

chrismcna, 2020-08-09:
Another random crash

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ede90  PS      : 0x00060330  A0      : 0x800d4c8c  A1      : 0x3ffb1f50
A2      : 0xabba1234  A3      : 0x3ffd4ca0  A4      : 0x3ffcef44  A5      : 0x3ff9cffa
A6      : 0x0000000a  A7      : 0x3ffcef48  A8      : 0x800edeb6  A9      : 0x3ffb1f20
A10     : 0xabba1234  A11     : 0x3ffb0000  A12     : 0x00000100  A13     : 0x3ffb367c
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xabba1248  LBEG    : 0x401df268  LEND    : 0x401df275  LCOUNT  : 0x00000000

Backtrace: 0x400ede90:0x3ffb1f50 0x400d4c89:0x3ffb1f70 0x400db459:0x3ffb1f90 0x4011a985:0x3ffb1fb0 0x4008d74d:0x3ffb1fd0
  #0  0x400ede90:0x3ffb1f50 in lv_task_handler at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_misc/lv_task.c:331
  #1  0x400d4c89:0x3ffb1f70 in gui_loop(TTGOClass*) at src\gui/gui.cpp:94
  #2  0x400db459:0x3ffb1f90 in loop() at C:/Users/Chris/Documents/GitHub/My-TTGO-Watch/src/my-ttgo-watch.ino:94
  #3  0x4011a985:0x3ffb1fb0 in loopTask(void*) at C:\Users\Chris\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:19
  #4  0x4008d74d:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

chrismcna, 2020-08-09:
Another

E (68315) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (68315) task_wdt:  - async_tcp (CPU 0/1)
E (68315) task_wdt: Tasks currently running:
E (68315) task_wdt: CPU 0: ipc0
E (68315) task_wdt: CPU 1: loopTask
E (68315) task_wdt: Aborting.
abort() was called at PC 0x4011c4d4 on core 0

Backtrace: 0x40091410:0x3ffbe170 0x40091641:0x3ffbe190 0x4011c4d4:0x3ffbe1b0 0x40087fb1:0x3ffbe1d0 0x40083971:0x3ffb9d60 0x4008b8d6:0x3ffb9d80 0x400863d3:0x3ffb9da0 0x4008d74d:0x3ffb9dc0
  #0  0x40091410:0x3ffbe170 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x40091641:0x3ffbe190 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x4011c4d4:0x3ffbe1b0 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40087fb1:0x3ffbe1d0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154 
  #4  0x40083971:0x3ffb9d60 in esp_intr_noniram_enable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c:297
  #5  0x4008b8d6:0x3ffb9d80 in spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:88
  #6  0x400863d3:0x3ffb9da0 in ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62
  #7  0x4008d74d:0x3ffb9dc0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

chrismcna, 2020-08-09:
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40090332  PS      : 0x00060034  A0      : 0x8008f4ef  A1      : 0x3ffbe650
A2      : 0x3ffb84c8  A3      : 0x3ffbc7f8  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060023  A7      : 0x00000000  A8      : 0x3ffbc7f8  A9      : 0x3ffbc7f8
A10     : 0x00000019  A11     : 0x00000019  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff
Core 1 was running in ISR context:
EPC1    : 0x4008e7c2  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40090332

Backtrace: 0x40090332:0x3ffbe650 0x4008f4ec:0x3ffbe670 0x4008d553:0x3ffbe690 0x4008d5bd:0x3ffbe6d0 0x40118659:0x3ffbe6f0 0x40117a01:0x3ffbe710 0x40117d09:0x3ffbe730 0x40081412:0x3ffbe780 0x40081485:0x3ffbe7b0 0x40087fb1:0x3ffbe7d0 0x401e0a97:0x3ffbc6e0 0x4011e226:0x3ffbc700 0x4008f165:0x3ffbc720 0x4008d74d:0x3ffbc740
  #0  0x40090332:0x3ffbe650 in vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:188
  #1  0x4008f4ec:0x3ffbe670 in vTaskPlaceOnEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #2  0x4008d553:0x3ffbe690 in xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1996 
  #3  0x4008d5bd:0x3ffbe6d0 in xQueueTakeMutexRecursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1996
  #4  0x40118659:0x3ffbe6f0 in i2cApbChangeCallback at C:\Users\Chris\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:463
  #5  0x40117a01:0x3ffbe710 in triggerApbChangeCallback at C:\Users\Chris\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-cpu.c:213
  #6  0x40117d09:0x3ffbe730 in setCpuFrequencyMhz at C:\Users\Chris\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-cpu.c:213
  #7  0x40081412:0x3ffbe780 in pmu_irq() at src\hardware/pmu.cpp:70
  #8  0x40081485:0x3ffbe7b0 in __onPinInterrupt at C:\Users\Chris\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-gpio.c:274
  #9  0x40087fb1:0x3ffbe7d0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154 
  #10 0x401e0a97:0x3ffbc6e0 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492     
  #11 0x4011e226:0x3ffbc700 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #12 0x4008f165:0x3ffbc720 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #13 0x4008d74d:0x3ffbc740 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Core 0 register dump:
PC      : 0x4008e7cd  PS      : 0x00060034  A0      : 0x8008f19d  A1      : 0x3ffbe150
A2      : 0x3ffbf9d8  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffcfae4
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x0000003f
A14     : 0x0000002a  A15     : 0x3ffe52c0  SAR     : 0x00000016  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x4008e7cd:0x3ffbe150 0x4008f19a:0x3ffbe180 0x4008d83b:0x3ffbe1a0 0x400901a9:0x3ffbe1c0 0x40087fba:0x3ffbe1d0 0x40167a97:0x3ffe5150 0x40168002:0x3ffe5210 0x401680b1:0x3ffe5230 0x40168178:0x3ffe5260 0x401666a5:0x3ffe52e0 0x401666e6:0x3ffe53a0 0x4016855d:0x3ffe53d0 0x401685f3:0x3ffe5440 0x40190051:0x3ffe5490 0x4019115a:0x3ffe54b0 0x40191169:0x3ffe54f0 0x401495f1:0x3ffe5510 0x40149a04:0x3ffe5550 0x4014a705:0x3ffe55c0 0x4014a7be:0x3ffe5610 0x40145c51:0x3ffe5660 0x40145eac:0x3ffe56b0 0x40145ec3:0x3ffe56f0 0x40148c30:0x3ffe5710 0x40148d5a:0x3ffe5730 0x40094553:0x3ffe5750 0x4008d74d:0x3ffe5790
  #0  0x4008e7cd:0x3ffbe150 in vPortCPUAcquireMutexIntsDisabledInternal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
      (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
      (inlined by) vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201
  #1  0x4008f19a:0x3ffbe180 in xTaskIncrementTick at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507   
  #2  0x4008d83b:0x3ffbe1a0 in xPortSysTickHandler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
  #3  0x400901a9:0x3ffbe1c0 in _frxt_timer_int at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:303     
[... cut ...]
  #20 0x40145c51:0x3ffe5660 in clear_bss_queue at ??:?
  #21 0x40145eac:0x3ffe56b0 in clear_bss_queue at ??:?
  #22 0x40145ec3:0x3ffe56f0 in scan_enter_oper_channel_process at ??:?
  #23 0x40148c30:0x3ffe5710 in wifi_station_stop at ??:?
  #24 0x40148d5a:0x3ffe5730 in ieee80211_timer_do_process at ??:?
  #25 0x40094553:0x3ffe5750 in ppTask at ??:?
  #26 0x4008d74d:0x3ffe5790 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

closed 2020-08-09 by sharandac

hutch120, 2021-02-04:
Just for reference, looks a bit like this issue... https://github.com/lvgl/lvgl/issues/1594

Workaround:

\My-TTGO-Watch\.pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\lvgl\src\lv_draw\lv_draw_label.c

Change

`if(txt[0] == '\0')  return;`

to

`if(txt == NULL || txt[0] == '\0')  return;`