sharandac/My-TTGO-Watch #34: Compiler Warnings: ISO C99 requires whitespace after the macro name

lienbacher, 2020-08-10:
I am a frequent platformio user and I have not had this issue before. I'm getting a multitude of identical compiler warnings when compiling: `<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name`

datacute, 2020-08-11:
This is due to the board name containing a hyphen "T-Watch" in both 
https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L4103
and
https://github.com/platformio/platform-espressif32/blob/master/boards/ttgo-t-watch.json#L7

I've changed mine to be "T_Watch".

Mine were installed to 
C:\Users\Stephen\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\boards.txt
and
C:\Users\Stephen\.platformio\platforms\espressif32\boards\ttgo-t-watch.json

lienbacher, 2020-08-11:
Ouch that one hurts! Thanks for that info @datacute!

sharandac, 2020-08-11:
@datacute Thanks for that info! I think it is also a good idea to tell the creator of the t-watch library that.

lienbacher, 2020-08-11:
@sharandac this triggers a thought. I understand the thought-process of the t-watch library, however it is just a library of libraries instead of just using dependecies like everyone else. it tries to make things simple but at the same time just makes it more complicated. Might be worth refactoring.

datacute, 2020-08-11:
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/issues/15

sharandac, 2020-08-11:
@datacute That's right. I'd rather not use the library at all. But it's an easy way to go these days. I suppose it will be in the future.

closed 2020-08-11 by sharandac