sharandac/My-TTGO-Watch #40: BLE communication using Gadgetbridge - Feature Request

alexkoppelman, 2020-08-11:
I would prefer BLE communication with my smartphone above WIFI (as I normally wear it outside), the project here: [https://github.com/bburky/t-watch-2020-project ](https://github.com/bburky/t-watch-2020-project ) has a pretty solid Gadgetbridge implementation (watch receiving messsage notifications) and functionality can be added  (play pause buttons for music, find my phone, ...)

sharandac, 2020-08-11:
I already saw this a few days ago and am already thinking about how it could be integrated.

sharandac, 2020-08-12:
I played a little with BLE. Unfortunately the additional RAM is very high (about 70kb) and the reconnect does not work. But I will take a closer look when I have enough time. That would be a nice feature.

JoanMCD, 2020-08-12:
before PSRAM the way was to shut down the webserver and start BT  - and vice versa.

closed 2020-08-14 by sharandac