sharandac/My-TTGO-Watch #41: Build error (Windows): The command line is too long.

tonhuisman, 2020-08-11:
After updating to the latest sources, since last Friday, and trying to upload, I get the error message `The command line is too long.`
Environment: Windows 10, VSCode & platformio (all with latest updates)
Tried the `build` and `upload` actions from the Pio menu in VSCode.

Moving the source folder closer to C:\ hasn't solved the issue.

NB: The OTA update of version 2020080702 doesn't seem to do anything...

tonhuisman, 2020-08-11:
Small update: After a quite long wait, and with no information whatsoever on the display (`start update...` isn't very informative, imho...), the phone rebooted and the update was successfully installed.
But that doesn't fix the build error.

alexkoppelman, 2020-08-12:
Have the same here. Moved the folder to the root of C:\ with no success. 
Environment: Windows 10, VSCode & platformio (all with latest updates)

cranphin, 2020-08-12:
Same for me, seems the linking command is too long
Might be related to: https://community.platformio.org/t/linker-the-command-line-is-too-long-esp32/8792/2 ?

pio run -v gives:
```
xtensa-esp32-elf-g++ -o .pio\build\ttgo-t-watch\firmware.elf -T esp32_out.ld -nostdlib -Wl,-static -u call_user_start_cpu0 -Wl,--undefined=uxTopUsedPriority -Wl,--gc-sections -Wl,-EL -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u __cxa_guard_dummy -u __cxx_fatal_exception .pio\build\ttgo-t-watch\src\app\example_app\example_app.cpp.o .pio\build\ttgo-t-watch\src\app\example_app\example_app_main.cpp.o .pio\build\ttgo-t-watch\src\app\example_app\example_app_setup.cpp.o .pio\build\ttgo-t-watch\src\app\example_app\images\example_app_48px.c.o .pio\build\ttgo-t-watch\src\app\example_app\images\example_app_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_01d_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_01n_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_02d_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_02n_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_03d_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_03n_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_04d_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_04n_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_09d_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_09n_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_10d_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_10n_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_11d_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_11n_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_13d_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_13n_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_50d_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\owm_50n_64px.c.o .pio\build\ttgo-t-watch\src\app\weather\images\resolve_owm_icon.cpp.o .pio\build\ttgo-t-watch\src\app\weather\weather.cpp.o .pio\build\ttgo-t-watch\src\app\weather\weather_fetch.cpp.o .pio\build\ttgo-t-watch\src\app\weather\weather_forecast.cpp.o .pio\build\ttgo-t-watch\src\app\weather\weather_setup.cpp.o .pio\build\ttgo-t-watch\src\gui\font\Ubuntu_16px.c.o .pio\build\ttgo-t-watch\src\gui\font\Ubuntu_72px.c.o .pio\build\ttgo-t-watch\src\gui\gui.cpp.o .pio\build\ttgo-t-watch\src\gui\images\battery_icon_64px.c.o .pio\build\ttgo-t-watch\src\gui\images\bluetooth_64px.c.o .pio\build\ttgo-t-watch\src\gui\images\brightness_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\brightness_64px.c.o .pio\build\ttgo-t-watch\src\gui\images\check_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\down_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\exit_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\foot_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_1_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_1_8px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_2_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_2_8px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_3_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_3_8px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_fail_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_fail_8px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_n_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_n_8px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_ok_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_ok_8px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_update_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\info_update_8px.c.o .pio\build\ttgo-t-watch\src\gui\images\lock_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\move_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\move_64px.c.o .pio\build\ttgo-t-watch\src\gui\images\refresh_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\setup_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\setup_64px.c.o .pio\build\ttgo-t-watch\src\gui\images\time_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\time_64px.c.o .pio\build\ttgo-t-watch\src\gui\images\trash_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\unlock_16px.c.o .pio\build\ttgo-t-watch\src\gui\images\up_32px.c.o .pio\build\ttgo-t-watch\src\gui\images\update_64px.c.o .pio\build\ttgo-t-watch\src\gui\images\wifi_64px.c.o .pio\build\ttgo-t-watch\src\gui\keyboard.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\app_tile\app_tile.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\main_tile\main_tile.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\mainbar.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\note_tile\note_tile.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\setup_tile\battery_settings\battery_settings.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\setup_tile\battery_settings\battery_view.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\setup_tile\display_settings\display_setting.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\setup_tile\move_settings\move_settings.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\setup_tile\setup.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\setup_tile\time_settings\time_settings.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\setup_tile\update\update.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\setup_tile\update\update_check_version.cpp.o .pio\build\ttgo-t-watch\src\gui\mainbar\setup_tile\update\update_setup.cpp.o .pio\build\ttgo-t-watch\src\g [... cut ...] rnet -limage_util -lc_nano -lsoc -ltcp_transport -lc -lmicro-ecc -lface_recognition -ljson -lwpa_supplicant -lmesh -lesp_https_ota -lwpa2 -lexpat -llwip -lwear_levelling -lapp_update -ldriver -lbt -lespnow -lcoap -lasio -lnewlib -lconsole -lapp_trace -lesp32-camera -lhal -lprotobuf-c -lsdmmc -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lwifi_provisioning -lwps -lesp_adc_cal -lesp_event -lopenssl -lesp_ringbuf -lfr -lstdc++ -Wl,--end-group
The command line is too long.
```

sharandac, 2020-08-12:
Thanks for the information! Unfortunately I can't confirm this, I don't have Windows to test it. Any ideas, anyone?

cranphin, 2020-08-12:
> 
> 
> Thanks for the information! Unfortunately I can't confirm this, I don't have Windows to test it. I think this is a problem which only exists under Windows. Any ideas?

@sharandac ,
This is a Windows issue yes.
The 'root' cause is that windows is more limited to command line length then unix alike systems.

That said, if you read the platformio issue I linked, you'll see that platformio is supposed to automatically apply a workaround for this issue on windows, by using a text file to provide arguments instead of directly via the command line.

Maybe that fix doesn't work (well), for example maybe there's a corner case where the workaround isn't triggered yet, but the line is already too long.
I've added a comment/link there back to here, maybe someone will pick it up there if we're lucky.

Alternatively, from your project there might be some fixes possible also.
If you look at my output, you'll see that it's the linker command line which becomes too long (~8k characters).
Anything that would shorten that line would fix the issue, maybe:
- Change paths to shorter or less deep ones (from the project root, the project location doesn't matter if you look at the line created).
- Have less files to link in the project (less c/source files).
- Somehow change parts to be linked together in one file that's linked in the final linking stage, in esp-idf I would use components for that, I'm not sure what options arduino/platformio has for that, other then making part a library.

tonhuisman, 2020-08-12:
As the information in the link @cranphin provides suggests, this is most likely a bug in platformio, as, how I read it there, pio is supposed to use a command-file if the command-line grows too long. Don't know if this is configurable from `platformio.ini`.

ondragu, 2020-08-12:
I´m having the same problem since a few days. Must be an platformio update issue. Before everything was fine.
Is there a different way to compile perhaps in arduino ?
Or can you post a .bin file ?

cranphin, 2020-08-12:
> 
> 
> I´m having the same problem since a few days. Must be an platformio update issue. Before everything was fine.
Just to be clear, I don't think it's a platformio update issue, but a longer existing issue which is triggered once a project has a lot of files to compile and link :)

sharandac, 2020-08-12:
@ondragu take a look in the repository, here you can find an actual ttgo-t-watch2020_v1.ino.bin file to upload to your device via webserver or directly with the build-in t-watch update ( setup/update )

grasshopper122, 2020-08-13:
@sharandac Tried uploading the .bin provided but the watch is unresponsive...

`C:\Users\User1>C:\Users\User1\Documents\Arduino\hardware\espressif\esp32/tools/esptool/esptool.exe --chip esp32 --port COM9 --baud 2000000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\User1\Downloads\ttgo-t-watch2020_v1.ino.bin`

FantasyFactory, 2020-08-13:
Hi to all !
I've simply  added a line into platformio.ini:
`lib_archive = false`

It worked for me

cranphin, 2020-08-13:
> 
> 
> @sharandac Tried uploading the .bin provided but the watch is unresponsive...
> 
> `C:\Users\User1>C:\Users\User1\Documents\Arduino\hardware\espressif\esp32/tools/esptool/esptool.exe --chip esp32 --port COM9 --baud 2000000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\User1\Downloads\ttgo-t-watch2020_v1.ino.bin`
> 
> I believe there should be a file like `ttgo-t-watch2020_v1.ino.partitions.bin` to go with it.

The bin only contains the app, you can flash it with esptool, but only if you send it to the right place.
What you've done now is overwrite the bootloader.
You'll have to reflash the watch with pio (use an older version of the project, or use the fix above if it works), to restore the bootloader.

The partitions currently used are (bootloader is not in the list, but I expect that's at 0x0):
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x640000,
app1,     app,  ota_1,   0x650000,0x640000,
spiffs,   data, spiffs,  0xc90000,0x370000,

So to flash the bin with esptool you'd have to flash it to 0x10000 (app0) or 0x650000 (app1), depending on which is the current active one.

sharandac, 2020-08-13:
@cranphin Yeah, you're right. The .bin file contains the pure app and must be written to the right place.

Can someone confirm that the fix works?

cranphin, 2020-08-13:
> 
> 
> @cranphin Yeah, you're right. The .bin file contains the pure app and must be written to the right place.
> 
> Can someone confirm that the fix works?

The option lib_archive = false also works for me as a workaround.
Looks like whatever it does, it also results in a much shorter linker command.
It's documented here: https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-archive
Reading it I'm not entirely sure what it does, but it doesn't sound overly dangerous :)

alexkoppelman, 2020-08-14:
did a new pull this morning and managed to build fine. Looks resolved

sharandac, 2020-08-14:
tkanks for your feedback!

closed 2020-08-14 by sharandac

tonhuisman, 2020-08-14:
Thanks guys!