sharandac/My-TTGO-Watch #42: Reboots

ondragu, 2020-08-14:
Since the last updates I am getting really much reboots. 
Seems to appear when the WLAN has connected. 
If I disable WiFi it is getting better.

sharandac, 2020-08-14:
Oh, no. In most cases ArduinoJson is involved and triggers the watchdog timer from the IDLE0 task. Could you test the latest version (2020081421) ? And if possible maybe record a debug output?

ondragu, 2020-08-15:
Still reboots when connecting to WiFi. sometimes better when I change to another screen before WiFi is connected.
How can I record a debug output in VS ?

sharandac, 2020-08-17:
If the watch is connected to VS, you can see the output in the Serial Monitor. A simple copy&paste from the moment where the watch crashes is enough.

ondragu, 2020-08-17:
starting t-watch V1, version: 2020081701
[W][SPIFFS.cpp:54] begin(): SPIFFS Already Mounted!
[I][pcf8563.cpp:332] syncToSystem(): syncToSystem: 2020 8 17 - 19 14 53

[I][pcf8563.cpp:342] syncToSystem(): syncToSystem: 120 8 17 
- 19 14 53

[I][bma.cpp:56] bma_setup(): stepcounter not valid. reset
[I][wifictl.cpp:334] wifictl_register_cb(): register wifictl_event_cb success
[I][blectl.cpp:327] blectl_register_cb(): register blectl_event_cb success
[I][blectl.cpp:327] blectl_register_cb(): register blectl_event_cb success
[I][wifictl.cpp:334] wifictl_register_cb(): register wifictl_event_cb success
[I][blectl.cpp:327] blectl_register_cb(): register blectl_event_cb success
[I][wifictl.cpp:334] wifictl_register_cb(): register wifictl_event_cb success
[I][wifictl.cpp:334] wifictl_register_cb(): register wifictl_event_cb success
[I][wifictl.cpp:334] wifictl_register_cb(): register wifictl_event_cb success
[I][wifictl.cpp:441] wifictl_on(): request wifictl on
[I][main_tile.cpp:163] main_tile_update_task(): renew time_str: 19:14 !=
[I][wifictl.cpp:343] wifictl_send_event_cb(): call wifictl_event_cb
[I][statusbar.cpp:203] statusbar_wifictl_event_cb(): wifictl event msg: scan ...
[I][BLEDevice.cpp:554] getAdvertising(): create advertising
[I][blectl.cpp:269] blectl_setup(): BLE advertising...      
Total heap: 233952
Free heap: 90572
Total PSRAM: 4176204
Free PSRAM: 3926828
[E][esp32-hal-misc.c:109] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
[I][blectl.cpp:336] blectl_send_event_cb(): call blectl_event_cb
[I][statusbar.cpp:193] statusbar_blectl_event_cb(): blectl event msg: connected
[I][blectl.cpp:65] onConnect(): BLE connected
[I][blectl.cpp:336] blectl_send_event_cb(): call blectl_event_cb
[I][statusbar.cpp:193] statusbar_blectl_event_cb(): blectl event msg: connected
[I][blectl.cpp:65] onConnect(): BLE connected
[I][BLEDevice.cpp:253] gapEventHandler(): ESP_GAP_BLE_AUTH_CMPL_EVT
[I][blectl.cpp:104] onAuthenticationComplete(): Bluetooth pairing successful
[I][blectl.cpp:183] onWrite(): attention, new link establish[I][blectl.cpp:186] onWrite(): attention, new message
[I][wifictl.cpp:343] wifictl_send_event_cb(): call wifictl_event_cb
[I][statusbar.cpp:203] statusbar_wifictl_event_cb(): wifictl event msg: connecting ...
[I][wifictl.cpp:526] wifictl_Task(): request wifictl on done[I][wifictl.cpp:343] wifictl_send_event_cb(): call wifictl_event_cb
[I][wifictl.cpp:343] wifictl_send_event_cb(): call wifictl_event_cb
[I][tidate.cpp:205] update_Task(): start update task        
␀task
[I][wifictl.cpp:343] wifictl_send_event_cb(): call wifictl_event_cb
[I][statusbar.cpp:203] statusbar_wifictl_event_cb(): wifictl event msg: HP4 192.168.188.31
[I][update.cpp:205] update_Task(): start update task        
[I][wifictl.cpp:343] wifictl_send_event_cb(): call wifictl_event_cb
[I][wifictl.cpp:343] wifictl_send_event_cb(): call wifictl_event_cb
[I][weather_forecast.cpp:194] weather_forecast_sync_Task(): 
start weather forecast task
[I][weather.cpp:176] weather_widget_sync_Task(): start weather widget task
[I][webserver.cpp:373] asyncwebserver_start(): enable webserver and ssdp
[I][blectl.cpp:188] onWrite(): message complete, fire BLTCTL_MSG callback
[I][blectl.cpp:196] onWrite(): msg: setTime(1597684497);E.setTimeZone(2.0);(s=>{s&&(s.timezone=2.0)&&require('Storage').write('setting.json',s);})(require('Storage').readJSON('setting.json',1))
[I][blectl.cpp:336] blectl_send_event_cb(): call blectl_event_cb
[E][bluetooth_call.cpp:103] bluetooth_call_msg_pharse(): update check deserializeJson() failed: InvalidInput
[I][main_tile.cpp:163] main_tile_update_task(): renew time_str: 21:14 != 19:14
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400013f9  PS      : 0x00060530  A0      : 0x80118e4d  A1      : 0x3fff0470
A2      : 0x3ffc7094  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x3ffc7094  A11     : 0x3f4fbe6d  A12     : 0x00000000  A13     : 0x3fff00c0
A14     : 0x3fff02c0  A15     : 0x00000001  SAR     : 0x00000012  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff

Backtrace: 0x400013f9:0x3fff0470 0x40118e4a:0x3fff0480 0x40119008:0x3fff07e0 0x400931fd:0x3fff0880
  #0  0x400013f9:0x3fff0470 in ?? ??:0
  #1  0x40118e4a:0x3fff0480 in weather_fetch_forecast(weather_config_t*, weather_forcast_t*) at src\app\weather/weather_fetch.cpp:117
  #2  0x40119008:0x3fff07e0 in weather_forecast_sync_Task(void*) at src\app\weather/weather_forecast.cpp:200
  #3  0x400931fd:0x3fff0880 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)   

Rebooting...
ets Jun  8 2016 00:22:57

sharandac, 2020-08-17:
thanks for your debug log. the last trace entry has disappeared, unfortunately. today i have rewrite some code in this section, so i hope i find this problem and resolve them. it is a little bit hard to reproduce this.

ondragu, 2020-08-17:
Strange. Other users seem not to have such problems. I have already erased the flash throug platformio.
What else can I do ?
The last update did not work for me either.
Greeting and thank a lot.

sharandac, 2020-08-17:
strange. an exciting question would be if the extended powersave is activated? there the core voltage is lowered to 3V. maybe switching it off helps if it is activated.

datacute, 2020-08-18:
@ondragu that looks like it has a problem when fetching the weather forecast. What response do you get from http://api.openweathermap.org/data/2.5/forecast?cnt=16&lat=%s&lon=%s&appid=%s
replacing %s with your lat/lon/appid?
See whether it helps to turn off your weather setting that says it should check when connected.
Check your lat & lon - it looks like the weather forecast doesn't include a city name if the lat & lon don't match to a city.

ondragu, 2020-08-25:
Was good until todays last update no. 2020082501.
Now it is again rebooting after 1 minute.

sharandac, 2020-08-25:
Wow ...I am irritated. No changes in hardware drivers for a few days. But what would be interesting: the heap is now quite full, could be a problem. Do you have a debug log?

closed 2020-08-26 by sharandac