sharandac/My-TTGO-Watch #43: Battery percentage is 10x the actual

5tormChild, 2020-08-14:
Two days before the battery percentage was just right after two charging cycles using calculated battery percentage. (97% is pretty close to what I call a full charge.)
But in one of the previous builds what was 70%, now it displays 592% instead of 70%:
"designed cap: 300mAh, charged capacity: 1776.0mAh. battery voltage: 3.68V"
If I turn calculated battery percentage off, it displays 17%, which is far from the reality.

sharandac, 2020-08-14:
please see setup->battery->setup and turn "calculated percent" off

5tormChild, 2020-08-14:
As I wrote, calculated percentage was just right two days earlier, but now
the remaining capacity is 10x the reality, that's why the percentage is off.
Turning the calculated percentage off makes the remaining battery
percentage off by a lot, like 16% instead of 70%.

sharandac, 2020-08-14:
I know. That's the problem with the power management unit. The measurement of the current in standby is unfortunately very inaccurate, so the error adds up. Both are very imprecise. I am not satisfied with it :(

sharandac, 2020-08-14:
Note: under 3.20V battery voltage, current counter reset to zero. like calibration.

datacute, 2020-08-15:
The calculation is 65536 * current LSB * (charge coulomb counter value - discharge coulomb counter value) / 3600 / ADC sample rate.
The sampling rate changed from 200Hz to 25Hz in version 2020081307, so the calculated percentage (and capacity) will be wrong till the battery voltage drops below 3.2V and the coulomb counter values are reset.

sharandac, 2020-08-17:
uh... my mistake. now I understand. It must have been the weekend :)
I'm gonna change it back. I wanted to see if I can save more energy in standby with this change. Unfortunately no. It will only be less accurate.

closed 2020-08-18 by sharandac

5tormChild, 2020-08-18:
It's s lot better now, thanks for your hard work on making the TTGO watch
usable!