sharandac/My-TTGO-Watch #48: Information Request: How to get started with

joshvito, 2020-08-17:
I'm an experienced web app coder, and want to get started working on a fork for this watch. 
I'd love to be able to contribute changes/features to this code base in the future.

I had a few questions about getting started.

1. Is there a explanation of how to use the web server on the watch to push code updates?
2. Also, is there a way to emulate the software running on the watch on my pc, to speed up development? I built and uploaded the software onto the watch and it took a bit.
3. Maybe a wiki page on here to explain theses things would be helpful? 
4. Any debugging tips? e.g. Can I attach a debugger to the software on the watch?

Thanks.

sharandac, 2020-08-17:
On the website of the watch you can upload a new image which has been created with platformIO via "updates" (http://x.x.x.x/upload.htm). But you can only update the whole watch and not single apps.
Unfortunately I haven't found a way to emulate the watch on the PC yet. But you can run LVGL on your PC for the first steps to get familiar with it.
Debugging is one such story. You would need an external debugger. But the Watch does not offer the possibility to connect it. As a pro tip: develop the basics of your app under arduino/platformIo without the watch on an another ESP32 module and then "port" it.This makes it easier to compile the app all the time.
Currently the API for the interface is not stable. A good starting point is /src/app/example_app/.
Unfortunately it is unavoidable that you should be familiar with C/C++ and heavy use of pointers to understand what is happening.

joshvito, 2020-08-17:
yea. Coming from a C# world, my initial thought was to get more familiar with C/C++ first.

closed 2020-08-20 by sharandac