sharandac/My-TTGO-Watch #49: Bluetooth problem

d3th3n, 2020-08-17:
There is no way to connect my watch to my iPhone SE 1st generation (just for testing connexion) is "normal" ?

sharandac, 2020-08-17:
Oh, too bad. Does the Iphone SE1 support BLE? I don't have an iPhone available to test it

d3th3n, 2020-08-17:
Yes it do ;) my hearing aids work with it... but I can't view my watch inside the Bluetooth device list... I can only view my watch with some ble scan app ! Probably because there no service available for my phone (like message/mail/ ring...)

sharandac, 2020-08-17:
i use gadgetbridge to send messages to the watch. no idea how it works with iOS and if there is something similar there. but in the end it is just a bluetooth-uart connection and the data is sent as json.

d3th3n, 2020-08-18:
just one question ... is anyone successful in getting bluetooth to work with an iphone?

closed 2020-08-20 by sharandac