sharandac/My-TTGO-Watch #51: Starter questions.

robouden, 2020-08-19:
Thanks all for the nice work on the atch2020 software.
I have some simple questions as a starer:

- How to change the background of the watch?
- Are there different themes available?

sharandac, 2020-08-19:
>     * How to change the background of the watch?
take a look at /src/gui/gui.cpp, line 48 and 60. https://lvgl.io/tools/imageconverter is your friend
>     * Are there different themes available?
no, i have no themes implemented

robouden, 2020-08-19:
Thanks for the reply. I will have a look.

regards,
rob

ondragu, 2020-08-19:
I am also looking forward to be able to change the background.
Black looks very good ;-) and perhaps it would reduce energy consumption.

[image]

robouden, 2020-08-19:
ondragu,

Did you already make the background like that, or you created the screenshot by other software?

regards
rob

sharandac, 2020-08-19:
you can take a screenshot with the built in webserver. gimp can read the file as raw, color format is 565. with an resolution of 240x240 pixel.

ondragu, 2020-08-19:
So did I.
It is a little bit tricky to open it with Gimp, had to rename it to .data.
Then choose the right format (was RGB Big Endian) and you have to say that it is 240x240 pixel.

sharandac, 2020-08-19:
a good hint to rename the file extension :)

datacute, 2020-08-20:
On windows, I can view the screenshots in irfanview. If the file extension is .raw then irfanview will ask what format to expect, and it remembers the settings for the next time.

FantasyFactory, 2020-08-20:
> I am also looking forward to be able to change the background.
/gui/gui.cpp
line 48 `LV_IMG_DECLARE(bg3)`
line 60 `    lv_img_set_src(img_bin, &bg3 );`
you can try the others defaul background simply changing bg3 in bg2, bg1 or bg

> Black looks very good ;-) and perhaps it would reduce energy consumption.
This is not OLED, so black pixel are no power saving

closed 2020-08-27 by sharandac

robouden, 2020-08-31:
I did some more testing and got finally my background the way I wanted. See  screenshot. 
[image]

BTW.  Enoying the interaction in this channel.

sharandac, 2020-08-31:
It is not a pro app, but I like hedgehogs :-)

In the future I wanted to build my own framework out of it which is not only suitable for the T-Watch-2020.

robouden, 2020-09-03:
Got the display even more to my liking with my company name in it. And the bootscreen just the background. 
(Sorry, Dirk I took out the Hedgedog, although I liked the design)

[image]
[image]