sharandac/My-TTGO-Watch #55: General questions

FantasyFactory, 2020-08-19:
Hi Dirk!
First, I want to thank you all for this great piece of software, which is an amazing starting point ... but a start to go where? I believe that everyone's desire could be to develop a useful framework.

Wish List: (not in order of importance or feasibility)
1) Bluetooth notification system: it is really the minimum for a smartwatch
2) a system for downloadable / installable "Apps". It is difficult to think of pre-compiled binaries linked at runtime, we would need a real operating system capable of dynamically allocating them, protecting memory, managing multitasking ... it's unrealistic, but a microscripting system would be interesting instead ! Micropython uses too many resources, but maybe an implementation of esp8266basic could allow the creation of small "apps" downloadable as source.
A further alternative could be to implement a "web app" system: a sort of lvgl dialog interpreter defined by a json readed via http would be enough, which could act as a sort of "form" whose result is sent back via http post. All the application logic would be processed server side, leaving ttwatch just to render the interface.
3) a more practical or at least more reliable way to manage the filesystem would be needed. Okay via http, the ideal would be a fat via usb.

What do you think ?

Personally, I would like to work on some "apps" (based on example_app to work in the current "framework")
1) a simple vnc client
2) a simple widget of a value (of your choice) taken from an OBDII wifi interface
3) one button to control some smart relays
4) time and days between now an a selectable date/time

sharandac, 2020-08-19:
Hey,

thank you! First of all I want to counteract the impression that I develop professional software for the Watch. It surprises me that so many wishes are brought to me (please do not take it personally). It is a hobby project for me. I have programmed a framework, so that others can use it for their purposes. If others find it great they can join in. It is open source.

1. I have programmed only the minimum what I wanted to have. Call signalling and message reading with [Gadgetbridge](https://gadgetbridge.org).
2. I think it is a little overkill to write an interpreter. The ESP32 is built for monolithic binary. But that's my opinion. But if someone wants to do it. I think the best way would be [ESP8266basic](http://www.esp8266basic.com). That's a lot of work.
3. The watch USB is only UART and not programmable, there is no way.

A VNC client sounds crazy! When? :)
OBDII? Are you a race car enthusiast?

cranphin, 2020-08-20:
> thank you! First of all I want to counteract the impression that I develop professional software for the Watch. It surprises me that so many wishes are brought to me (please do not take it [...]
Your firmware is almost the only, and by far the best available firmware. AND it's installed on new watches, so no wonder people think you're the official developer ;)

> OBDII? Are you a race car enthusiast?
I've got an old smelly car and I use ODB, it's nice to see what your engine is up to :) It's just a serial protocol, so might not be too hard (but different for each car), But I'm not sure I'd recommend it on watch, looking at your watch while driving is definatly a distraction one doesn't need (dash mounted phone is much better for it) :)

sharandac, 2020-08-20:
:-)

https://forum.micropython.org/viewtopic.php?f=18&t=8642

FantasyFactory, 2020-08-20:
> I've got an old smelly car and I use ODB, it's nice to see what your engine is up to :) It's just a serial protocol, so might not be too hard (but different for each car), But I'm not sure I'd [...]

the intention was to hook the clock to the steering wheel, however the clock could also act only as a "data logger" perhaps adding the accelerometer values, but more than anything else it was for the fun of developing the "app" for the ttwatch , on mobile phones there are already very good apps (such as Torque Pro)

FantasyFactory, 2020-08-20:
I have some other ideas, but I think I would be the only one to use certain apps: D

I use a Sony mirrorless camera that exposes interesting APIs via wifi, in particular I often use "vintage" lenses and my problem then during the editing phase is to remember which lens I used for which photos, in order to apply the relative calibration profile.
I would be very comfortable having an app on the clock that synchronizes the time with the camera and has a simple menu in which I set which lens I am using, producing a "log" that tells me "11:23 Helios 44 lens; 11:31 Zeiss Planar 50 lens; 11:40 xxx lens etc "

Another very stupid App would be the clock at 25 hours: D: D I set an hour of the day that must necessarily correspond with the real time, for example 8, and another "twenty-fifth" hour to have for example between 19 and 20: in this way at 8 the clock would mark 8, but it would go slightly ahead until 18:30 in which it would indicate 19, then the clock would remain STOPPED until 19:31 showing however 19:01 and so on until recovering and re-synchronizing at 8: D: D every hour would therefore last 57.5 minutes, a negligible difference, and I would have a "twenty-fifth hour" to dedicate to what I never have time to do :D (yes I'm totally mad)

d3th3n, 2020-08-20:
For Gadgetbridge, what configuration (watch name) did you use to communicate with the T-Watch?

sharandac, 2020-08-20:
"Espruino (T-Watch2020)" is the name. so Gadgetbridge can find it.

d3th3n, 2020-08-20:
thank you, not having Android but only Apple ... I am looking for a replacement "solution" (gadgetbridge not existing on Apple)

FantasyFactory, 2020-08-20:
Gadgetbridge works great with twatch !

sharandac, 2020-08-20:
Nice. At the moment I still want to integrate [OsmAnd](https://osmand.net/).

d3th3n, 2020-08-20:
After a lot of research, I found a way about BLE and IOS notifications: the ANCS also called "Apple Notification Service" (another Apple protocol?) ... one way to use Apple services directly  on the watch without third-party software (Android too?)

sharandac, 2020-08-20:
look [here](https://www.espruino.com/Gadgetbridge), this is what gadgetbridge sends via BLE as json

alexkoppelman, 2020-08-20:
> look [here](https://www.espruino.com/Gadgetbridge), this is what gadgetbridge sends via BLE as json

Using this, I am trying to get data/commands from the watch back to GadgetBridge.  

I've tried both (summarizing): sending a message from watch to phone (hopefully later some activity data)
```
       String msg1;
        DynamicJsonDocument doc(1024);
        doc["t"] = "music";
        doc["n"] = "Play";
        serializeJson(doc, msg1);
        //unsigned int lastStringLength = msg1.length();
        pTxCharacteristic->setValue(msg1.c_str()); // Set value.
        pTxCharacteristic->notify();                // Notify value.
```

and try to play/pause music player:

        String mydata = "{t:\"music\",n:\"play\"}";
        pTxCharacteristic->setValue(mydata.c_str()); // Set value.
        pTxCharacteristic->notify();                // Notify value.

And I can see my phone is receiving as I get the string through in the TX Characteristic of the UART Service when I look with the android nRFC Connect app. However, gadgetbridge does not seem to like it as it does no respond in any way. 

I am wondering; Is this still a work in progress no the GadgetBRidge Side or am I doing something wrong? 
Any ideas?

sharandac, 2020-08-20:
I am not sure, but setValue() expects a double and no string.It looks like you have to split your data.
I also have to read the documentation first.

sharandac, 2020-08-23:
@helgetnt Which music player do you use? My favourite music player (DSub) does not support this.

helgetnt, 2020-08-23:
@sharandac 
I was thinking of an app that works like the buttons on a headset regardless of the player used so it can be used on Android & iOS. ......... PS .: I've only just seen that I have to use the Gadgetbridge app on Android. However, I have not yet found the function for sending BT messages.
I just saw that you are from Germany too. :-)

MfG HelgeTNT

helgetnt, 2020-08-23:
I've found out at bluetooth.cpp: from line 118
                if ( doc["number"] ) {
                    **_if ( doc["name"] ) {
                        lv_label_set_text( bluetooth_call_number_label, doc["name"] );
                    }
                    else {
                        lv_label_set_text( bluetooth_call_number_label, doc["number"] );
                    }
                }_**
                else {
                    lv_label_set_text( bluetooth_call_number_label, "" );
                }
...so it shows the name( if it's in the phonebook ), else it shows the number.

MfG HelgeTNT

helgetnt, 2020-08-23:
.............sorry that's in bluetooth_call.cpp .......................
MfG

robouden, 2020-08-26:
Silly question. But where is the nice grey background image stored in the code? I like to set up the main background like that. (maybe a bit darker).
[image]

sharandac, 2020-08-26:
This is not a silly question. The grey is not a background image. If you comment out the lines 55-60 in the last version of gui.cpp you see what I mean. After that you can play with the opacity in the file mainbar.cpp in line 56.

`lv_style_set_bg_opa(&mainbar_style, LV_OBJ_PART_MAIN, LV_OPA_0);`

to

`lv_style_set_bg_opa(&mainbar_style, LV_OBJ_PART_MAIN, LV_OPA_100);`

robouden, 2020-08-26:
Thanks for the quick reply.

Did commented out the lines as you suggested. Seems to work ok. Only one issue if I set the lv_style_set_bg_opa(&mainbar_style, LV_OBJ_PART_MAIN, LV_OPA_0); to lv_style_set_bg_opa(&mainbar_style, LV_OBJ_PART_MAIN, LV_OPA_20); A message remains on the screen. See picture.
[image]

robouden, 2020-08-26:
Got the message issue easy fixed with adding:
    splash_screen_stage_update( "", 80 );
after the:
   splash_screen_stage_update( "init gui", 80 );
[image]

sharandac, 2020-08-26:
In the latest version it is fixed. the splash routine will not remove all labels after splash :)

robouden, 2020-08-26:
Dirk,

I assume you mean "now remove" instead of " not remove"?

Regards,
Rob Oudendijk.

sharandac, 2020-08-26:
i mean the old splash routine will not remove all labels ... now remove

closed 2020-08-27 by sharandac