sharandac/My-TTGO-Watch #62: App guidelines

FantasyFactory, 2020-08-21:
starting from stopwatch and weather I'm trying to develop my first twatch "app";

Apps (APPNAME, replace it with what you like) must be included in main file my-ttgo-watch.ino
thru #include "app/APPNAME/APPNAME.h" 
then adding a setup function APPNAME_setup(); in the arduino setup() function

the APPNAME_setup( void ) function basically will create the gui thru the lvgl library, read the json config file, then wil do some other initialization

#ifdef APPNAME_WIDGET will enable some other functions (more than mere icon show in main tile, but I've not yet understood)

This is the little I understand :D 

Events handling are due by callback functions. Most of them are related to lvgl, but which "system events"or hooks  we can handle ? For example, what I have to do to handle a custom message not handled by  bluetooth_message_msg_pharse() or override it's handling ?

So, there is some visual editor for the lvgl gui  ? 
What are the guidelines for developing an App that correctly interact whith the framework "My-ttgo-watch" ?

FantasyFactory, 2020-09-25:
despite the little time I have to devote to this beautiful toy, I am writing my first "app" :D it is a simple application that **reads from the web an image**, such as in my case an image coming from a surveillance camera. In this way I'm also **trying to write a guide** that might be useful to others who want to try to write some simple app for this watch. Unfortunately at the moment I can only read via http **but NOT https**, if anyone can help me... regarding the image format, I had initially attempted a porting of **TJpgDec**, but at the moment I have dropped the jpgs and I am content with the png that seem simpler and less exhite than ram. I implemented the **lodepng** library with a few minor modifications, and its custom decoder for lvgl.

merlinludwig, 2021-01-16:
I have had this watch for a little over 24 hours.
I am rusty because I was never a programmer despite taking a few classes and I haven't touched any kind of code in a few years.
I pouring over this sourcecode and wondering all the same things.
I'm *almost* tempted to make a massive visual diagram of everything just so I know exactly what's what and where and how to hook into it all.

jasongayda, 2021-03-02:
> So, there is some visual editor for the lvgl gui ?

@FantasyFactory : You can use this application to create GUI for the watch.
https://www.nxp.com/design/software/development-software/gui-guider:GUI-GUIDER
It will even export code for you to use in your app.

jasongayda, 2021-03-02:
> despite the little time I have to devote to this beautiful toy, I am writing my first "app" :D it is a simple application that **reads from the web an image**, such as in my case an image coming [...]

@FantasyFactory : you can use this tool for image conversion: https://lvgl.io/tools/imageconverter

drunkscientist, 2021-03-11:
so just today lvgl guys released a beta version of the ui editor, they call it edgeline, [here is a video](https://www.youtube.com/watch?v=UrSkzbuuGaw&ab_channel=LVGL), [here is a link to it](https://forum.lvgl.io/t/v0-1-beta-is-out/4818), its currently windows only and missing most of the widgets still but i thought id share