sharandac/My-TTGO-Watch #71: Error in build

gosymbian, 2020-08-24:
Hi,

I have loaded the project into PlatformIO and when I try to build it, I get the following :

> Executing task: C:\Users\Dominique HUGO\.platformio\penv\Scripts\platformio.exe run <

Processing ttgo-t-watch (platform: espressif32; board: ttgo-t-watch; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
WindowsError: [Error 183] Cannot create a file when that file already exists: 'C:\\DISK_C\\_OneDrive\\Arduino\\Lilygo\\My-TTGO-Watch-master\\.pio\\build\\ttgo-t-watch':
  File "C:\users\dominique hugo\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 155:
    makedirs(env.subst("$BUILD_DIR"))
  File "c:\users\dominique hugo\.platformio\penv\lib\os.py", line 157:
    mkdir(name, mode)
============================================= [FAILED] Took 181.81 seconds =============================================
The terminal process "C:\Users\Dominique HUGO\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
Terminal will be reused by tasks, press any key to close it.

I have change nothing in code, same .ini downloaded from github
can you advice please ?
thanks in advance,
sincerly,
Domi.

chrismcna, 2020-08-24:
Hmm, very strange, 
the header is located here for me "C:\Users\Chris\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src"

can you check to see if you have the header under the a similar location

chrismcna, 2020-08-24:
C:\Users\Chris\.platformio\packages\framework-arduinoespressif32\.piopm 
says im on "version": "3.10004.200129"

robouden, 2020-08-24:
The error shows wire.h missing seems strange to me. Did you install the espressif32 platform?

regards,
rob

gosymbian, 2020-08-24:
> The error shows wire.h missing seems strange to me. Did you install the espressif32 platform?
> 
> regards,
> rob

Hi rob,
yeps, espressif32 is installed.

[image]

thanks,
Domi.

gosymbian, 2020-08-24:
> Hmm, very strange,
> the header is located here for me "C:\Users\Chris\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src"
> 
> can you check to see if you have the header under a similar location

Hi Chris,

yes, I have the files :

[image]

thanks,
Domi.

gosymbian, 2020-08-24:
> C:\Users\Chris.platformio\packages\framework-arduinoespressif32\.piopm
> says im on "version": "3.10004.200129"

Hi Chris,

I haven't this file ...

[image]

sincerly,
Domi.

gosymbian, 2020-08-24:
I have re-started from scratch, delete all the directories, ... and everything works fine now ! I can compile and upload the project.

Thanks to all of you for your time & help,
sincerly,
Domi.

closed 2020-08-24 by sharandac