sharandac/My-TTGO-Watch #73: Time zone has no effect

gosymbian, 2020-08-24:
Hi,

Time zone has no effect for me, I have tried +2, +3, it always stays on the same hour ...

any clue ?
thanks,
Domi.

sharandac, 2020-08-24:
After timesync over wifi it show the right time.

gosymbian, 2020-08-24:
my router doesn't send the time ....

Domi.

vanvuongngo, 2020-08-24:
Hi Domi,
when I change the time zone it has not immediately  an effect. A timesync after a value change would be pretty. I have to switch off and on again to invoke a timesync.

sharandac, 2020-08-24:
Please enable timesync like here

[image]

after next wifi connect the time will sync automaticly.

closed 2020-08-26 by sharandac

robouden, 2020-08-26:
Silly question. But where is the nice grey background image stored in the code? I like to set up the main background like that. (maybe a bit darker).