sharandac/My-TTGO-Watch #74: Hard Code Wifi credentials

alosbh, 2020-08-26:
Hello, 

First of all i wanna say congratz for the great firmware you have developed. Im a starter but it is definitely a good starting point. Amazing work!

I'm used to Arduino general settings of Wifi... Just provide ssid and psk and do WiFi.Begin... But as a beginner , the wifictl code seems a little tricky for me :( 
Can you help me hard code my credentials to instead of start scanning, connect to my SSID ?
 I added WiFi.Begin(ssid,password) to wifictl.cpp wifictl_on(), but it didnt worked as expected, since there are some other functions from this wifictl module that conflicts with WiFi.

Thanks

vanvuongngo, 2020-08-26:
You can setup a `data/wificfg.json` but I did not figure out how the format have to be. The watch restarts infinitely.

alosbh, 2020-08-26:
> You can setup a `data/wificfg.json` but I did not figure out how the format have to be. The watch restarts infinitely.

Thanks @vanvuongngo. My thoughts are that it must be something like wpa_supplicant for Linux... I saw it defined at wifictl.h, but without a sketch I dont think I can figure it out too...

sharandac, 2020-08-26:
i think the best way is to do it through the wifi setup after scanning. there you can enter the password directly after clicking on your SSID. or use WPS :)

[image]

alosbh, 2020-08-26:
Thanks @sharandac ! Surely the wifi setup would work for discoverable ssid. There is something i can do to connect to a hidden ssid? Thanks !

sharandac, 2020-08-26:
One way is, add

`wifictl_insert_network("ssid","password");`

at the of the wifictl_setup routine.

sharandac, 2020-08-26:
Or you can force it by add

`WiFi.begin( wifiname, wifipassword );`

between  line 107 and 108

closed 2020-08-27 by sharandac

alosbh, 2020-08-28:
Hi guys, 
was able to connect using WiFi.begin over WiFi.scanNetworks() on wifi event callback of SYSTEM_EVENT_WIFI_READY

[image]

Just remember to do the same in callback of SYSTEM_EVENT_STA_DISCONNECTED.
So when you miss the connection, it will try to reconnect to your hardcoded SSID again, instead of scanning networks.

Thanks