sharandac/My-TTGO-Watch #75: Crypto ticker & note question

jcwren, 2020-08-27:
Playing around with the various built-in stuff in the latest release. Nice bit of work you've done here!

I've entered BTC for the crypto ticker symbol, but never get an update. Anything I'm missing here?

The Note function doesn't seem to do anything. I haven't dived into the code yet, so maybe that's just a dummy page.

Got the weather update working, had to retype the API key a couple times. That wasn't any fun.

robouden, 2020-08-27:
It  maybe would have been easier for you maybe to use the webserver and eneter the API key there I think. See attached screenshot on the edit page .
[image]

jcwren, 2020-08-27:
Ah, cool. I knew it had the web server built in, but I haven't tried that
yet. I for sure didn't know you could edit with it. I was only aware of the
/shot path for screenshots. Might be good to mention some of this early in
the README.md file.

--jc

Save a life; adopt a shelter animal.

jcwren, 2020-08-27:
If your SSID is too long, the IP address is being clipped. In my case, it's "Keyhole 12 Spy Satellite", and the IP address is 172.16.3.1xx, but the 'xx' is clipped. I know what it is from watching my DHCP server logs, but that's not an easy option for everyone.

sharandac, 2020-08-27:
Please update to the last version: 2020082701

robouden, 2020-08-27:
Updated. Runs fine here.
[image]

robouden, 2020-08-27:
At taking a screenshot it crashed once. After that no issues. Maybe cache issue?
Screen looks nice now. Still I would like to find a way to make a sold grey background.

[image]

jcwren, 2020-08-29:
Still have a problem with the crypto ticker not working on the latest version. I get this error:
```
[I][crypto_ticker_main.cpp:212] crypto_ticker_main_sync_Task(): start crypto ticker main task, heap: 27064
[I][statusbar.cpp:220] statusbar_wifictl_event_cb(): statusbar wifictl event 0002, msg: 172.16.3.131
[I][webserver.cpp:374] asyncwebserver_start(): enable webserver and ssdp
[I][weather.cpp:214] weather_widget_sync_Task(): finish weather widget task, heap: 16904
[E][crypto_ticker_fetch.cpp:100] crypto_ticker_fetch_statistics(): crypto_ticker deserializeJson() failed: InvalidInput
[I][crypto_ticker_main.cpp:244] crypto_ticker_main_sync_Task(): finish crypto ticker main task, heap: 23548
```

Thijxx, 2021-01-29:
Please re-open. Crypto ticker does not work because no service at URL.

sharandac, 2021-01-29:
@chrismcna Any information about this issue?

Thijxx, 2021-01-29:
`#define MY_TTGO_WATCH_HOST    "my-ttgo-watch.co.uk"` in `crypto_ticker_fetch.h` seems to be an invalid source of the ticker data, maybe if you guide me I can test with a working market API from Binance or something

sharandac, 2021-01-30:
@Thijxx this domain and app is from @chrismcna . i don't know if he's reading this to fix it.

closed 2021-02-07 by sharandac