sharandac/My-TTGO-Watch #76: Which version of TWatch are you using?

lewisxhe, 2020-08-27:
Hi, I tried to update and compile the latest firmware, and found that the alignment of characters in the menu has changed. I don’t know which version of TWatch lib you are using. Can you provide a commit id? I have updated lvgl to version 7.3.1 in the TWatch lib, but I think it has nothing to do with this, because the alignment of the original example is correct!

lewisxhe, 2020-08-27:
[image]
[image]

lewisxhe, 2020-08-27:
Because Twatch lib needs to be compatible with multiple boards, I can only try to ensure compatibility with this project. Thank you for the example you wrote for twatch, it is great

sharandac, 2020-08-27:
Hey, i use the last release 1.2.0. it looks like the definitions are 

`LV_HOR_RES_MAX`

and

`LV_VER_RES_MAX`

are wrong

sharandac, 2020-08-27:
And I saw that they are planning a new watch 2020 v2. While you are at it, a 32kHz crystal would be great. see here: 

[https://github.com/espressif/esp-idf/issues/947](https://github.com/espressif/esp-idf/issues/947)

This would make it possible to use BLE for a longer period of time while the watch is asleep and save energy.

lewisxhe, 2020-08-27:
Well, I will suggest adding a 32KHZ crystal! As for the resolution, that is the ST7796 screen size I set it to fit, so you should use lv_disp_get_hor_res() and lv_disp_get_ver_res()

sharandac, 2020-08-27:
Thanks for the hint! i will adapt. Can you test it?

lewisxhe, 2020-08-27:
[image]

Ok, I can submit a request to you if needed

sharandac, 2020-08-27:
looks good. not necessary. i have already changed the corresponding lines in the code.

sharandac, 2020-08-27:
And another small hardware bug. the covers on the back breaks after a while. all at the same place, near the cutout of the speaker. i think they should be made a bit stronger.

[image]

lewisxhe, 2020-08-27:
Okay, I have given feedback!

sharandac, 2020-08-27:
I have problems with the new version of TWatch lib. By default DMA is enabled. I have moved the screen memory to the psram to have more internal ram. This causes the dma transfer to crash.

lewisxhe, 2020-08-27:
I will provide a concise way for LVGL instead of using TFT_eSPI as a driver, because for using LVGL, using TFT_eSPI will undoubtedly consume more ROM, please wait for a while, I will update

lewisxhe, 2020-08-27:
Please provide your email, we will give you a free watch after testing the 2020V2 version.

sharandac, 2020-08-27:
A simple way is when use

`#define TWATCH_USE_PSRAM_ALLOC_LVGL`

to disable 

`#define ENABLE_LVGL_FLUSH_DMA`

sharandac, 2020-08-27:
> Please provide your email, we will give you a free watch after testing the 2020V2 version.

THX! dirk.brosswick@googlemail.com

lewisxhe, 2020-08-27:
Ok ! If using DMA causes instability, please disable DMA first

sharandac, 2020-08-27:
This is not possible. Line 104 in TTGO.h force this by default.

robouden, 2020-08-27:
Lewsx He,

If you need another tester for T-Watch200V2 let me know. I like to test too!!

regards,
Rob Oudendijk

lewisxhe, 2020-08-27:
I will disable DMA first. In fact, using DMA and normal transfer, I feel the effect is almost the same!

sharandac, 2020-08-27:
Me to! I have change the line 104 from

`#if !defined(EXTERNAL_TFT_ESPI_LIBRARY) && !defined(LILYGO_BLOCK_ILI9488_MODULE)`

to

`#if !defined(EXTERNAL_TFT_ESPI_LIBRARY) && !defined(LILYGO_BLOCK_ILI9488_MODULE) && !defined(TWATCH_USE_PSRAM_ALLOC_LVGL)`

now it works fine for me.

datacute, 2020-08-27:
Does a 32kHz clock signal work for low power BLE too?
https://github.com/espressif/esp-idf/issues/947#issuecomment-511264081
 The RTC can supply that can't it? The relevant ESP32 pin is used for something else though currently.

sharandac, 2020-08-27:
I'm not sure. this was just an idea to increase the runtime with ble, currently about 10-12 hours. but the arduino environment does not support this at the moment. dynamic frequency scaling (DFS) is not enabled. I guess we will have to live with that.

lewisxhe, 2020-08-27:
@robouden Then you also tell me your email, we will give you a free test, hope you can help us develop the firmware

lewisxhe, 2020-08-27:
@sharandac Twatch lib has been updated, I will update the lvgl dedicated display function later

robouden, 2020-08-27:
Lewisx He

My email address is rob@yr-design.biz.

regards,
Rob Oudendijk

sharandac, 2020-08-27:
@lewisxhe Thanks a lot!

apo76, 2020-08-27:
@lewisxhe : If you need one more beta tester, I'm in! ;o)
titanium2k4@hotmail.com .
Regards,
Attilio.

lewisxhe, 2020-08-27:
We will send to you after the test, and email to contact you before sending @robouden  @sharandac 

@apo76  Sorry, there are no more free places! Sorry

closed 2020-08-27 by lewisxhe

apo76, 2020-08-27:
@sharandac : I installed Gadgetbridge app after reading one of your post, but if I want notifications of missed phone call or messages on my watch, I need to set the BLE always on and I have a maximum autonomy of half a day, which is clearly not usable....is it the same on your side?
Best Regards,
Attilio.

sharandac, 2020-08-27:
Yes, this is a problem with ESP32 and BLE. Unfortunately this cannot be changed :(

cranphin, 2020-08-27:
> 
> 
> Yes, this is a problem with ESP32 and BLE. Unfortunately this cannot be changed :(

It would probably require the hardware update (32khz crystal), and using the esp32-idf, which means letting go of arduino environment.. :)
See: https://github.com/espressif/esp-idf/issues/947, not sure if the software side of that ever got implemented correctly either.

I tried the same thing with Gadgetbridge, and it does eat up the battery. It would be better if gadgetbridge would send any missed notification when the watch (re)connects on bluetooth, you wouldn't have live notifications, but at least you could check by waking the watch.
Setting autoreconnect in gadgetbridge works for me (the connecting part), but no missed notifications :)

cranphin, 2020-08-28:
> Okay, I have given feedback!

@lewisxhe About the cracking shell, I see a very slight crack only on the outside, in the same place as the picture, without the shell having been under much pressure. I wonder if it's thick enough, but: A. there was a problem with the plastic molding, causing stress, or B. with the design it's just a tiny bit too tight in the case, so there's always a bit too much pressure on that side.

NorthernDIY, 2020-09-02:
@lewisxhe If you are still able to take suggestions for v2, a light sensor and a status led might be pretty useful additions.

If your board layout guy is feeling generous, even just breaking out a few of the unused IO pins on the ESP32 would be super helpful for any mods we might want to install.

I've actually dug into the hardware and added my own mods on GPIO2 (LED) and GPI34 (Light sensor).  There is enough space to mount them along side the IR transmitter.  I haven't developed any code to support it just yet aside from basic testing to make sure they work.