sharandac/My-TTGO-Watch #77: Using it with T-Watch 2019

alosbh, 2020-08-28:
Hello guys, 

Has any of you managed to get this firmware working properly on T-Watch 2019? 

I tried to `#define LILYGO_WATCH_2019_WITH_TOUCH` but yet the touchscreen doesnt seems to work. Main screen is displayed, looks like all other functionalities booted properly. But the screen cant recognize my touch.

Switching only this line of code above worked fine for lilygo examples library (https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/)

Any thoughts?

Thanks!

wlcims, 2020-08-28:
The touch screens of T-Watch2019 and T-Watch2020 are reversed, you can modify it by modifying `src/hardware/touch.cpp`
```diff
diff --git a/src/hardware/touch.cpp b/src/hardware/touch.cpp
index b5209c7..c6ec7da 100644
--- a/src/hardware/touch.cpp
+++ b/src/hardware/touch.cpp
@@ -58,31 +58,36 @@ static bool touch_getXY( int16_t &x, int16_t &y ) {
 
     p = ttgo->touch->getPoint();
 
+    p.x=(p.x-4)*240/(320-4);
+    p.y=(p.y-20)*240/(320-20);
+
     uint8_t rotation = ttgo->tft->getRotation();
     switch ( rotation ) {
-    case 0:
+    case 2:
         x = TFT_WIDTH - p.x;
         y = TFT_HEIGHT - p.y;
         break;
-    case 1:
+    case 3:
         x = TFT_WIDTH - p.y;
         y = p.x;
         break;
-    case 3:
+    case 1:
         x = p.y;
         y = TFT_HEIGHT - p.x;
         break;
-    case 2:
+    case 0:
     default:
         x = p.x;
         y = p.y;
     }
 
     // issue https://github.com/sharandac/My-TTGO-Watch/issues/18 fix
-    float temp_x = ( x - ( lv_disp_get_hor_res( NULL ) / 2 ) ) * 1.15;
-    float temp_y = ( y - ( lv_disp_get_ver_res( NULL ) / 2 ) ) * 1.0;
-    x = temp_x + ( lv_disp_get_hor_res( NULL ) / 2 );
-    y = temp_y + ( lv_disp_get_ver_res( NULL ) / 2 );
+   // float temp_x = ( x - ( LV_HOR_RES_MAX / 2 ) ) * 1.0;
+   // float temp_y = ( y - ( LV_VER_RES_MAX / 2 ) ) * 1.0;
+   // x = temp_x + ( LV_HOR_RES_MAX / 2 );
+   // y = temp_y + ( LV_VER_RES_MAX / 2 );
+
+   log_i("touch x:%d, y:%d",x,y);
 
     return( true );
```

alosbh, 2020-08-28:
Thank you @wlcims !! Worked like a charm!

Best regards;

closed 2020-08-28 by alosbh