sharandac/My-TTGO-Watch #78: Webserver not starting

datacute, 2020-08-29:
What causes asyncserver.begin() to report bind error -8?
```
[E][AsyncTCP.cpp:1268] begin(): bind error: -8
[I][webserver.cpp:374] asyncwebserver_start(): enable webserver and ssdp
```
Turning on the watch shortly after it has turned off (either through button press or timeout), and then clicking on the wifi symbol, or the 'turn wifi on' switch, I **sometimes** can't connect to the web server, and when monitoring the serial output, see that I've got this error.

datacute, 2020-08-29:
lwip\err.h
```
/** Address in use.          */
  ERR_USE        = -8,
```

sharandac, 2020-08-29:
Which firmware version do you use? Is it reproducible? And if so, how?

datacute, 2020-08-29:
That was with 2020082802, modified slightly to add some more battery details into the /info page.
Not reproducible every time.
With wifi set to not start automatically, I was repeatedly:

1. either turning the watch off or letting the watch time out and turn off
2. turning the watch on
3. starting wifi
4. opening a new chrome browser window to show the /info page.

It would fail to connect after roughly half a dozen repeats.
I've just updated to the (downloaded/unchanged) 2020082901 and will try again.

sharandac, 2020-08-29:
Most of the time I also have problems to reach the clock by ping for example although it is online. It gets the weather, updates and so on. But ping is not working. I think it is a general network problem. I have the same on other project here.

datacute, 2020-08-29:
Same experience with 2020082901 downloaded from you.
This also happens with just turning wifi off and on and refreshing the /info page.

closed 2020-09-03 by sharandac