sharandac/My-TTGO-Watch #83: Wifi scan doesn't complete

remcoder, 2020-09-02:
Scanning for wifi access points doesn't work for me. The scan doesn't seem to ever finish. Maybe it's because the stand-by mode turns off the wifi scan after 10s of inactivity. 

Reproduce:
- pull down the status bar
- tap on the Wifi icon. The Wifi icon turns green and the text 'scan...' is shown
... wait
- screen goes dark

- when I tap the screen or press the button on the side of the watch, the screen goes back on and it still appears to be scanning.
... until the screen goes dark again

hardware: TTGO 2020 v1
version: 2020090202 
tested commits 6a521af66a05565561ca19959aca0735c0650e15 and 4b35eb38a9c9c85d94048173e7db54cce8a3f13a

here a log output:
```
[I][wifictl.cpp:429] wifictl_off(): request wifictl off
[I][wifictl.cpp:500] wifictl_Task(): request wifictl off done
[I][webserver.cpp:380] asyncwebserver_end(): disable webserver and ssdp
[I][wifictl.cpp:334] wifictl_send_event_cb(): call wifictl_event_cb (0x40124304)
[I][statusbar.cpp:235] statusbar_wifictl_event_cb(): statusbar wifictl event 0010, msg: 
[I][wifictl.cpp:334] wifictl_send_event_cb(): call wifictl_event_cb (0x4011d4f0)
[I][weather_forecast.cpp:151] weather_forecast_wifictl_event_cb(): weather forecast wifictl event: 0010
[I][wifictl.cpp:334] wifictl_send_event_cb(): call wifictl_event_cb (0x4011bbb4)
[I][weather.cpp:88] weather_widget_wifictl_event_cb(): weather widget wifictl event: 0010
[I][wifictl.cpp:334] wifictl_send_event_cb(): call wifictl_event_cb (0x4011a2ec)
[I][wifictl.cpp:417] wifictl_on(): request wifictl on
[I][wifictl.cpp:334] wifictl_send_event_cb(): call wifictl_event_cb (0x40124304)
[I][statusbar.cpp:235] statusbar_wifictl_event_cb(): statusbar wifictl event 0008, msg: scan ...
[I][wifictl.cpp:505] wifictl_Task(): request wifictl on done
[I][main_tile.cpp:194] main_tile_update_task(): renew time_str: 05:46 != 05:45
[I][mainbar.cpp:200] mainbar_jump_to_tilenumber(): jump to tile 0 from tile 0
[I][powermgm.cpp:121] powermgm_loop(): Free heap: 81244
[I][powermgm.cpp:122] powermgm_loop(): Free PSRAM heap: 3906260
[I][powermgm.cpp:123] powermgm_loop(): uptime: 5452
[I][display.cpp:75] display_standby(): go standby
[I][bma.cpp:74] bma_standby(): go standby
[I][pmu.cpp:79] pmu_standby(): enable silence wakeup timer, 3min
[I][pmu.cpp:93] pmu_standby(): go standby, enable 3000mV standby voltage
[I][wifictl.cpp:444] wifictl_standby(): request wifictl standby
[I][wifictl.cpp:429] wifictl_off(): request wifictl off
[I][wifictl.cpp:500] wifictl_Task(): request wifictl off done
[I][webserver.cpp:380] asyncwebserver_end(): disable webserver and ssdp
[I][wifictl.cpp:334] wifictl_send_event_cb(): call wifictl_event_cb (0x40124304)
[I][statusbar.cpp:235] statusbar_wifictl_event_cb(): statusbar wifictl event 0010, msg: 
[I][wifictl.cpp:334] wifictl_send_event_cb(): call wifictl_event_cb (0x4011d4f0)
[I][weather_forecast.cpp:151] weather_forecast_wifictl_event_cb(): weather forecast wifictl event: 0010
[I][wifictl.cpp:334] wifictl_send_event_cb(): call wifictl_event_cb (0x4011bbb4)
[I][weather.cpp:88] weather_widget_wifictl_event_cb(): weather widget wifictl event: 0010
[I][wifictl.cpp:334] wifictl_send_event_cb(): call wifictl_event_cb (0x4011a2ec)
[I][wifictl.cpp:449] wifictl_standby(): request wifictl standby done
[I][powermgm.cpp:141] powermgm_loop(): go standby
```

sharandac, 2020-09-03:
Thanks for the report! Unfortunately I cannot understand the behaviour. Are other networks generally visible in the settings or does the list remain empty?

remcoder, 2020-09-03:
Hi @sharandac, I never see a list. Not even an emty one. The Wifi icon and the text 'scan...' are all that I see.

sharandac, 2020-09-03:
Then please swipe into the setup tile ( swipe right to the appication tile and  then down to the setup tile ) and take a look into the wifi settings. There you can enter your network credits after a short scan.

remcoder, 2020-09-03:
ah! I found the setup tile, found the list and connected to my wifi. Thank you ;-)

closed 2020-09-03 by remcoder

sharandac, 2020-09-03:
very gladly!