sharandac/My-TTGO-Watch #85: Sending Data to GadgetBridge

alexkoppelman, 2020-09-03:
I am trying to make a small app to control the music player on my android.

I have tried several ways of getting the [required json string to Gadgetbridge](https://www.espruino.com/Gadgetbridge) but none seem to work. They are received correctly, as I can see in the nRF Connect app, however Gadgetbridge ignores the commands.

```
`void ble_send_command(){
    log_i("Play clicked");
    
     // char* jsonString = (char*)"{ \"t\":\" music\", \"n\": \"play\" }";

    DynamicJsonDocument doc(1024);
    String play = "{ \"t\":\" music\", \"n\": \"play\" }";
    deserializeJson(doc, play);
    JsonObject obj = doc.as<JsonObject>();
    
    char buffer[1024];

    // this gives an error in gadgetbridge on malformed JSON:
    // serializeJsonPretty(obj, buffer);
    
    serializeJson(obj, buffer);
    pTxCharacteristic->setValue(buffer); // Set value.

    pTxCharacteristic->notify();                // Notify value.
}`
```

Using  serializeJsonPretty I get a malformed json error, but using  serializeJson the app only receives the first 20-or-so characters of the (correct looking) json string. I've tried adjusting the BLE MTU to a higher value, having no effect.

My question would be:
1. Does anyone have an idea if there is a buffer length in the BLE communications going out? How can I make sure the whole json string is sent?
2. Did anyone successfully communicate from watch to GadgetBridge in any way?

This functionality (sending commands to Gadgetbridge) would come in really handy; control music player, pickup/end calls when using a headset and have your phone in a bag;)

sharandac, 2020-09-03:
I think this is the same problem as receiving messages. The chuck size is always max 20Bytes. When receiving these chucks have to be reassembled. And don't forget, there are still control characters. The same will probably be done for sending.
This sounds like the next task :-)

alexkoppelman, 2020-09-03:
:) in the meantime, I'll be looking at the code from the [Espruino Bangleapps](https://github.com/espruino/BangleApps/blob/master/apps/gbridge/widget.js) to see if I can figure out how they resolved that.

alexkoppelman, 2020-09-03:
I've added the BLEDevice MTU setting right after the creation of the BLE Server (line 252-ish in blectl.cpp). Now at least in nRF Connect the complete json string is received (without manually setting the MTU). 

Gadgetbridge still ignores it though.

    ```
// Create the BLE Server
    pServer = BLEDevice::createServer();
    pServer->setCallbacks( new BleCtlServerCallbacks() );

    // SETTING HIGHER MTU
    uint16_t mtu = 517;
    BLEDevice::setMTU(mtu);
```

sharandac, 2020-09-03:
and when your message start with binary data like

`0x03, 0x0a`

and ends with

`0x10`

?

alexkoppelman, 2020-09-09:
Finally made some progress here. To send data back to GadgetBridge:

```
void ble_send_command(){
    char* jsonString3 = (char*)"\x0a{t:\"music\",n:\"play\"}\x10";
    pTxCharacteristic->setValue( jsonString3 );
    pTxCharacteristic->notify();
}
```

Command reference [here](https://www.espruino.com/Gadgetbridge).
I do still have to set the mtu higher (517) using nRF connect. This will be the next one to figure out as:

```
    uint16_t mtu = 512;
    BLEDevice::setMTU(mtu);
```

does not seem to work

sharandac, 2020-09-10:
i'll check it out tomorrow. i already have an idea. which player do you use for testing?

alexkoppelman, 2020-09-10:
Ace! The final command line that (intermittently) works  is:
`"\x0a{t:\"music\",n:\"pause\"}\x10"`

I'm using Google Music, for testing. Once that works I'll see if my strange "Smart Audiobook player" works when default is selected.

sharandac, 2020-09-11:
done! please test the experimental branch with this function like:

`void blectl_send_msg( char *msg );`

alexkoppelman, 2020-09-14:
looking good! Thanks for that!
For most of the actions, the button needs to be clicked twice though...

sharandac, 2020-09-14:
then just exchange 0x0a and 0x10 :) for testing

closed 2020-10-08 by sharandac