sharandac/My-TTGO-Watch #90: the bin file is not work. --ttgo-t-watch2020_v1.ino.bin

James0329, 2020-09-04:
I using ESP32 DOWNLOAD TOOL V3.8.5 try to flash your file to update my watch.
It doesn't work.
My settings is :
SPI 40MHZ 
SPI MODE DIO
FLASH SIZE 128Mbit
Flash address  0x0000

After flash done.Watch auto reboot by ESP32 DOWNLOAD TOOL order.
Screen doesn't work.Click key nothing happen.

NorthernDIY, 2020-09-04:
I think this a repeat of issue #79 
[Yesterday's issue](https://github.com/sharandac/My-TTGO-Watch/issues/79#issuecomment-683408413)

James0329, 2020-09-04:
I try what you said.
But not work.
I use esp32 download or esptool.exe both not work with this bin file.
Set address to 0x1000 is useless.
I try to use TTGO Watch lib in Arduino.
It is work well.
It can light up watch with a simple digital clock.
So my hardware , usbline,  OS env full functional and work well.
I pretty sure the problem is trigger by bin file.

James0329, 2020-09-04:
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 10:52:1c:65:4c:a0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 16384.2 kbit/s)...
Hash of data verified.
Flash params set to 0x024f
Compressed 15856 bytes to 10276...
Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.1 seconds (effective 1585.6 kbit/s)...
Hash of data verified.
Compressed 401824 bytes to 181430...
Wrote 401824 bytes (181430 compressed) at 0x00010000 in 3.8 seconds (effective 848.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 4915.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Here is the log by arduino upload process.
You can see there baud rate is 2000000 not 460800.
Wrote 8192 bytes (47 compressed) at 0x0000e000 
Wrote 15856 bytes (10276 compressed) at 0x00001000
Wrote 401824 bytes (181430 compressed) at 0x00010000
Wrote 3072 bytes (129 compressed) at 0x00008000

And we can see the arduino wrote program at 4 address.
Not only 0x1000 one address.
So does it mean the bin file ignore 3 parts of file to write?
Because of a uncomplete wrote effect TTGO-WATCH not work?

James0329, 2020-09-04:
btw I find some other esp32 program have a env.bin need to flash before app.
we follow the readme.md erase watch does it erase env?
if this bin file not include env.
I think it is the problem.

cranphin, 2020-09-05:
The problem is the bin file is only the application. You also need the bootloader, and possibly some other data/partitions.
So you MUST use platformio at least once to get all the parts on your watch.
After that you can use esptool.exe to flash the bin to 0x10000 to replace/update the existing app, though it's a bit of a hack (and you might have to use 0x650000 instead, depending on whether ota_0 or ota_1 is in use).

So short answer, use platformio to initially flash your watch, it's the only way that's gonna work.

James0329, 2020-09-05:
Guys.I solve it by myself.
Now I can tell you how to use this bin with ESP32 DOWNLOAD TOOL.
Ask your device seller to get a factory bin file.
Start ESP32 DOWNLOAD TOOL 3.8.5
1.Flash factory.bin to you watch at address 0x0000.
(Dont flash two bin files at same time.)
2.Flash twatch 2020 file .Change address to 0x10000

Tool Settings
SPI SPEED 40MHZ  / SPI MODE DIO / FLASH SIZE 128Mbit/Baud 460800

After flash done ，hold main key more than 2 sec then waiting screen on.
(btw,this bin file will flip your screen 180 degree ,so you need goto watch display settings to change it back.)

wlcims, 2020-09-05:
Refer to this
[https://github.com/wlcims/My-TTGO-Watch-Bin-Flash](https://github.com/wlcims/My-TTGO-Watch-Bin-Flash)

closed 2020-09-06 by sharandac