sharandac/My-TTGO-Watch #91: IRremote addition breaks my build

datacute, 2020-09-05:
The latest merge of the IRController app results in a build failure for me when trying to link.
```
...
Compiling .pio\build\ttgo-t-watch\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\ttgo-t-watch\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\ttgo-t-watch\libFrameworkArduino.a
Linking .pio\build\ttgo-t-watch\firmware.elf
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\ttgo-t-watch\firmware.elf] Error 1
================================================================================== [FAILED] Took 99.46 seconds ==================================================================================
The terminal process "C:\Users\Stephen\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

The IRremoteESP8266 library is included in the build, e.g.
`Compiling .pio\build\ttgo-t-watch\lib808\IRremoteESP8266\IRac.cpp.o`

NorthernDIY, 2020-09-05:
I just tried recompiling from a fresh clone of the repository in a new folder.  It seems to compile fine for me, but my build environment is Ubuntu based.

I don't have any real suggestions other than maybe antivirus related interference?

datacute, 2020-09-05:
Reverting to a previous commits works fine. I'm keen on building on your IR app.

NorthernDIY, 2020-09-05:
I still don't know the actual cause of the issue, but i'm glad the revert worked.

As a side note, be sure to keep the IR gpio low when not transmitting, there's no current limiting and it has a transistor driver so it gets HOT fast!

FantasyFactory, 2020-09-05:
If you are using PlaftormIO for VSCode, you could try to CLEAN the previous build
[image]
Worked for me

datacute, 2020-09-05:
Clean didn't work for me.
I did finally find a solution: I had to delete the IRController_setup.cpp file.

datacute, 2020-09-05:
Oh - deleting the IRController_setup.cpp file was the last thing that resulted in it finally linking. I first had to delete all the unused ir_Whatever and IRac files.
Why does my linker complain when there are some completely unused .o files?

NorthernDIY, 2020-09-05:
I just tried compiling on a Win 10 machine and it has the same problem you describe.  It is interesting that you had to delete some IRac files as afaik they are completely unused in the program.

The IRController_setup.cpp file is also unused but was included in the push as I had plans to add a tilt to resend feature you could enable in setup.

I tried removing both IRController_setup.h, and IRController_setup.cpp as well as removing the erroneous .h include in the main cpp file and neither of those did the trick.

I also tried enabling compiler optimizations -O2, but that didn't help either.

I'm not sure what the actual solution is here, I don't typically do development on Windows.

datacute, 2020-09-05:
Thanks for trying that and confirming that it's not unique to something just on my PC.
I wonder if it's because the IR library supports ESP8266, and my platformio installation has not had support for that board added (as all my ESP8266 development had so far been in Arduino IDE or VSCode with Arduino extension - currently disabled).

sharandac, 2020-09-06:
#94

datacute, 2020-09-06:
Removing the lib_archive line worked for me. (Platformio version 5.0.0)
Adding these build flags then reduced the size:
	-D_IR_ENABLE_DEFAULT_=false
	-DSEND_RAW=true

sharandac, 2020-09-06:
the two settings have no real effect on the size:

with additional build flags
```
RAM:   [==        ]  22.7% (used 74224 bytes from 327680 bytes)
Flash: [=====     ]  52.2% (used 3420137 bytes from 6553600 bytes)
```
without additional build flags
```
RAM:   [==        ]  22.7% (used 74224 bytes from 327680 bytes)
Flash: [=====     ]  52.3% (used 3428509 bytes from 6553600 bytes)
```
I think that would be unnecessary.

closed 2020-09-06 by sharandac

datacute, 2020-09-07:
I discovered IRremoteESP8266's suggestion to disable all but what you use, when I was looking for the cause of the linking failure. I've been doing lots of programming for ATtiny85 chips - where 8k would be 100% of flash, rather than 0.1% which I agree is not significant.