sharandac/My-TTGO-Watch #92: event callbacks for power management

jakub-vesely, 2020-09-05:
Dear Dirk,

I have time to continue on alarm clock app today. I'm following your idea regarding to event driven behavior. Unfortunately this mechanism is not fully implemented in power management. It is not possible to register callback functions. 

I took a look how it is implemented for rtcctl and it seems that there is already very 3 similar functions for registering callback functions. 

From my point of view, there are three possibilities how solve the issue:
* to use direct call to stop alarm when the device is going to standby 
* to copy/paste code from one of already implemented places and adjust it for needs of power management.
* to refactor the code to be used one base (template class?)

Which way do you prefer? Do you have another idea how to solve the issue?

sharandac, 2020-09-06:
Hey Jakub,

i think the second way is the best.

> to copy/paste code from one of already implemented places and adjust it for needs of power management.

Should I do that?

sharandac, 2020-09-06:
I have implemented callback now. see powermgm.h

jakub-vesely, 2020-09-06:
It works. Thank you!

closed 2020-09-06 by jakub-vesely

sharandac, 2020-09-06:
great! i am looking forward to the alarm clock.