sharandac/My-TTGO-Watch #94: error: CreateProcess: No such file or directory

jakub-vesely, 2020-09-06:
Master branch is not buildable. The build is terminated with the error  _xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory_. 
The issue appears with the commit f776a31caa0a9c4b0cdab8cd5dd2962b387b576b and its merge to master.

sharandac, 2020-09-06:
can you please test if it builds under windows 10 if

`lib_archive = true`

in platformio.ini

is set? here it works under windows 10 and xubuntu 20.04.

robouden, 2020-09-06:
Works fine on Linux Mint (Debian based) and PlatformIO 5.0.0 too.
[image]

jakub-vesely, 2020-09-06:
It seems that the commit 904dd2fbe11339183518ea694571a11c96ff9c39 solved the issue.  I'm able to build binary again. 
Thank you!

closed 2020-09-06 by jakub-vesely