sharandac/My-TTGO-Watch #95: convert pio project to arduino IDE project ?

picpic020960, 2020-09-06:
Bonjour ,
could someone do that?
thanks in advance
Regards

ua6ahp, 2020-09-10:
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library - it is a "Arduino-ready" library. Read how to configure Arduino IDE on the above page.
SimpeWatch can be loaded from examples - TTGO TWatch Library -> LVGL -> SimpleWatch. 

Before uploading, you may need to uncomment the line

**//#define LILYGO_WATCH_2020_V1 // To use T-Watch2020 , please uncomment this line**

in config.h file.

But there will be no additional applications in the firmware. But the interface is more convenient there.

jakub-vesely, 2020-09-15:
Hi, 
My I ask you what is the motivation for this request? I am just curious about that because I can't to see any advantage .
Thank you, Jakub

ua6ahp, 2020-09-16:
Not everyone is ready to get started with platformio. Therefore, they ask ...
But this project is a little tricky for arduino ide ...

closed 2020-09-25 by sharandac