sharandac/My-TTGO-Watch #100: Chat group for this project

ssspeq, 2020-09-09:
I was wondering if there is some kind of chatgroup or something along those lines for this project? Telegram, irc, something like that? If not, perhaps one could be created?

ua6ahp, 2020-09-10:
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library - it is a "Arduino-ready" library. SimpeWatch can be loaded from examples - TTGO TWatch Library -> LVGL -> SimpleWatch.  Read how to configure on the above page.

Before uploading, you may need to uncomment the line
//#define LILYGO_WATCH_2020_V1                  // To use T-Watch2020 , please uncomment this line
in config.h file.

But there will be no additional applications in the firmware. But the interface is more convenient there.

datacute, 2020-09-11:
Gitter.io allows signing in with github account, and creating groups for your github repositories.

ssspeq, 2020-09-12:
I took the liberty of creating a telegram group if anyone is interested.

https://t.me/TTGO_Watch

closed 2020-09-22 by sharandac