sharandac/My-TTGO-Watch #106: Error receiving bluetooth messages

datacute, 2020-09-12:
I was using version 2020091102 and attempting to have GadgetBridge set up an alarm.
Having no success, I took a look at the logs: 
```
[I][blectl.cpp:184] onWrite(): attention, new message
[I][blectl.cpp:186] onWrite(): message complete, fire BLTCTL_MSG callback
[I][blectl.cpp:188] onWrite(): gadgetbridge message identified, cut down to json
[I][blectl.cpp:190] onWrite(): msg: {"t":"alarm","d":[{"h":13,"m":25}]}
[I][blectl.cpp:369] blectl_send_event_cb(): call blectl_event_cb (0x400e01a0)
[E][wlan_settings.cpp:430] bluetooth_message_msg_pharse(): bluetooth message deserializeJson() failed: NoMemory
[I][blectl.cpp:369] blectl_send_event_cb(): call blectl_event_cb (0x400dbaa8)
[E][bluetooth_call.cpp:103] bluetooth_call_msg_pharse(): bluetooth call deserializeJson() failed: NoMemory
[I][blectl.cpp:369] blectl_send_event_cb(): call blectl_event_cb (0x400dc044)
[I][bluetooth_message.cpp:170] bluetooth_message_msg_pharse(): msg: {"t":"alarm","d":[{"h":13,"m":25}]}
[E][bluetooth_message.cpp:176] bluetooth_message_msg_pharse(): bluetooth message deserializeJson() failed: NoMemory
[I][blectl.cpp:369] blectl_send_event_cb(): call blectl_event_cb (0x400d9d10)
[E][weather_setup.cpp:276] bluetooth_message_msg_pharse(): bluetooth message deserializeJson() failed: NoMemory
[I][blectl.cpp:369] blectl_send_event_cb(): call blectl_event_cb (0x400d5ff4)
[I][powermgm.cpp:143] powermgm_loop(): Free heap: 135740
[I][powermgm.cpp:144] powermgm_loop(): Free PSRAM heap: 3882416
[I][powermgm.cpp:145] powermgm_loop(): uptime: 189
[I][display.cpp:75] display_standby(): go standby
```
"pharse" is a typo?
I don't know why they're reporting NoMemory.
Is it expected that the bluetooth message should be processed by multiple handlers?
Is it possible for the message to be converted to a json doc once, and that doc given to the various handlers (in addition to the raw message - which might not be json), or does that make it difficult to free the doc once it is no longer needed?

datacute, 2020-09-12:
https://arduinojson.org/v6/assistant/ helps explain... that message takes 1 array, and 2 objects of size 2, which is already 80 bytes on ESP32, (plus 14 bytes for string duplication?). The message length x 2 is only 70

sharandac, 2020-09-12:
Thanks for the hint. Calculating the size of a Json for unknown strings is a bit of magic. See here:

[determine-the-capacity-of-the-jsondocument](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)

Yes, each handler should evaluate the message itself. For the simple reason to have a separation between driver and app. The driver should only forward the message without evaluating the content.

closed 2020-09-13 by sharandac