sharandac/My-TTGO-Watch #112: wav

jakub-vesely, 2020-09-15:
I tried to play snd_signal.h and snd_start.h via sound_play_progmem_wav().
Sound is enabled in sound settings and the volume set to maximum. 
I verified that message "playing audio (size %d) from PROGMEM" is present in log.
Unfortunately, I still  can't to hear anything.  

Shall it work or is sound playing  still under development?

Thank you, Jakub

sharandac, 2020-09-16:
i haven't tested it myself. but the snd_start.h is not a wav file. it's a raw file. but i just noticed that the speech module doesn't work properly.

rnisthal, 2020-09-16:
I just tested it, I uploaded a file alarm.mp3 using the web server and set it to play after boot with sound_play_spiffs_mp3("/alarm.mp3");

Also I added sound_speak("Hi there, sound test");

And in both cases it played with very bad quality.

sharandac, 2020-09-16:
Oh, thanks for your comment and hint!

rnisthal, 2020-09-17:
I'm trying again today, and I'm able to make the watch speak, but so far mp3 playback is not working

rnisthal, 2020-09-17:
I put:

    log_i("Playing wakeup sound. Sound.enabled: %d", sound_get_enabled_config());
    sound_play_spiffs_mp3("/alarm.mp3");

in sound_wakeup.

When I wake up manually by pressing the crown the sound does not play, but from time to time when the watch wakes up by itself without waking the screen it plays the mp3, somewhat distorted.

Maybe an issue with ram? Or would it be better to make sound_loop a lvgl task or an OS task?

closed 2020-09-22 by sharandac