sharandac/My-TTGO-Watch #114: Battery calculations, callbacks and interrupts

datacute, 2020-09-19:
I think that the documented calculation to use with the coulomb counters, wrongly assumes 100% efficiency in the battery. If you measure all the energy going into the battery during charging, and all the energy coming out of the battery during discharging, you would expect to have to put more energy in than is able to be taken out through powering our watch.

I thought I'd create an event log, capturing the charge & discharge values, and other interesting details, whenever events happen. Then over time we can see the relationship between the coulomb counters and % battery left.

I've written some code to append to an event log file every time callback_send is called (in your experimental branch), and have noticed a couple of strange behaviours that have got me wondering whether my logging code is being interrupted...

1. Sometimes the charge coulomb counter and/or discharge coulomb counter values are returned as 2684354560 or 4026531840 (values that are more interesting in hex) instead of expected values of other log messages of 4535, and sometimes the battery % is read as 0. Charging and discharging currents also once returned strange values of 2031.5mA and 4063.5mA, and the battery voltage once read as 0.02V instead of 4.21V
2. The log includes a timestamp, and a few times, always with wifictl event WIFICTL_CONNECT_IP, the timestamp is 12 hours ahead of what it should be (my timezone is currently UTC+12, and the logged timestamp is essentially UTC+24)

Is there something I should do to protect the region of code that is writing the log line?

datacute, 2020-09-22:
Attached are logs from my watch, and a spreadsheet where I removed lines that had bad readings, converted temperatures, and added some graphs.
[watch_event_log.xlsx](https://github.com/sharandac/My-TTGO-Watch/files/5259347/watch_event_log.xlsx)
[event_log_2020-09-20.log](https://github.com/sharandac/My-TTGO-Watch/files/5259349/event_log_2020-09-20.log)
[event_log_2020-09-21.log](https://github.com/sharandac/My-TTGO-Watch/files/5259350/event_log_2020-09-21.log)
[event_log_2020-09-22.log](https://github.com/sharandac/My-TTGO-Watch/files/5259351/event_log_2020-09-22.log)

datacute, 2020-09-22:
The first few lines in the log files show the unusual date & time received.

At 2020-09-20 23:32:16 the coulomb counters got reset, but the battery voltage was above 3.2V, and discharge counter was well above the charge counter. What must have happened is a bad reading for the discharge counter, putting it above the charge counter, and triggering the code in our library that resets the counters.

datacute, 2020-09-22:
I suspect that these bad readings might be caused by not checking the return codes of I2C methods.

sharandac, 2020-09-22:
ah ... from time to time i see [i2c busy] messages. but i dont know where it comes. i think here is the problem.

datacute, 2020-09-22:
To investigate my suspicion, I've added this to the end of TTGO TWatch Library's i2c_bus.cpp readBytes method
```
	if (_port->lastError() != 0) {
		log_e("i2c error %s", _port->getErrorText(_port->lastError()));
	}
```
They're certainly not uncommon. (Including "i2c error OK" meaning I need to capture the error before giving the semaphore back.

closed 2020-09-25 by sharandac