sharandac/My-TTGO-Watch #119: What's the thinking behind "silence wakeup"?

datacute, 2020-09-22:
I can see that it turns the device on regularly, and therefore apps can be written to subscribe to those events, or perform loop functions when woken silently.
However I don't understand why the device stays on for the duration of the display timeout.
For example, if the display timeout is 2 minutes, the device repeatedly wakes for 2 minutes, then sleeps for 3 minutes.

NorthernDIY, 2020-09-22:
This sounds like a bug?

Pmu.h has SILENCEWAKEUPTIME defined as 45 (seconds) as a default value but I am not sure where it is used or how the end of the silent wake is triggered.

datacute, 2020-09-22:
That's 45 minutes not seconds, and SILENCEWAKEUPTIME_PLUG is 3 minutes, both defining how long it stays asleep for, not how long it wakes up for.
I must have been looking at the behaviour in the serial port, so plugged in, and sleeping for 3 minutes.

sharandac, 2020-09-22:
Yes, you're right. There are two times. One when the watch is plugged in and one when the watch is "on the move". Both times are in minutes and can also be set in the battery.json. Or they can be switched off completely.

NorthernDIY, 2020-09-22:
Perhaps we should rename this to SILENCEWAKETIMEINTERVAL, and add a new item for the silent wake timeout, or we could have tasks put into some sort of queue and have it go to sleep once all tasks are complete?

sharandac, 2020-09-22:
That's right. I'll make some adjustments to make that clearer.

closed 2020-09-25 by sharandac