sharandac/My-TTGO-Watch #120: No WIFI scan result even after 5 minutes

techboycr, 2020-09-24:
Hello!

Thanks for this project, amazing work.

I just upload it but Wifi is not wosing any results, i am doing it from the wifi option in the settings menu.

Any help? Any change I should do to the code?

Thanks in advance.

sharandac, 2020-09-24:
Oh ... it sound like an hardware bug ... what you see in the wifi setup menu? After a short scan all visible networks are shown here. with an tap on the network name you can enter your password here.

[image]

yourchanges, 2020-09-24:
same problems. 

nothing in the visible networks

techboycr, 2020-09-24:
@sharandac in my case where you have the list of networks I have a white space. I will get my logs and share them with you asap.

rnisthal, 2020-09-24:
I saw the same behaviour, no scan until you configure the wifi credentials using http://www.my-ttgo-watch.co.uk/ and bluetooth
To reproduce, delete wifi configuration file like if the watch were new and try to scan.
Then enter wifi credentials using bluetooth and from there on wifi scan works.

sharandac, 2020-09-24:
I can confirm the issue. please update to 2020092401.

geekahertz, 2020-09-25:
Had the same problem before but for me it is because the "data" folder is missing a bunch of JSON files.  
For Wifi config, it is  /data/wificfg.json
``
{
    "autoon": true,
    "webserver": false,
    "networklist": [
        {
            "ssid": "YOUR_SSID",
            "psk": "YOUR_PASSPHRASE"
        }
    ]
}
``

yourchanges, 2020-09-25:
> Had the same problem before but for me it is because the "data" folder is missing a bunch of JSON files.
> For Wifi config, it is /data/wificfg.json
> `{ "autoon": true, "webserver": false, "networklist": [ { "ssid": "YOUR_SSID", "psk": "YOUR_PASSPHRASE" } ] }`
> 
> ps. I upload to SPIFFS via platformio terminal
> `platformio run -e ttgo-t-watch -t uploadfs`

After upload wificfg.json , the device reboot  , here is the log 

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
starting t-watch V1, version: 2020092401 core: 1
Configure watchdog to 30s: 0
[I][callback.cpp:44] callback_init(): init callback_t structure success for: powermgm
[I][callback.cpp:95] callback_register(): register callback_func for powermgm success (0x400e5e54:display)
[I][callback.cpp:44] callback_init(): init callback_t structure success for: powermgm loop
[I][callback.cpp:95] callback_register(): register callback_func for powermgm loop success (0x400e5cf0:display loop)
[W][SPIFFS.cpp:54] begin(): SPIFFS Already Mounted!
[I][pmu.cpp:42] pmu_setup(): set target voltage to 4.36V
[I][callback.cpp:95] callback_register(): register callback_func for powermgm success (0x400e6fe4:pmu)
[I][callback.cpp:95] callback_register(): register callback_func for powermgm loop success (0x400e7224:pmu loop)
[I][callback.cpp:95] callback_register(): register callback_func for powermgm success (0x400e4f5c:bma)
[I][callback.cpp:95] callback_register(): register callback_func for powermgm loop success (0x400e5118:bma loop)
[E][callback.cpp:159] callback_send(): no callback structure found
[E][callback.cpp:159] callback_send(): no callback structure found
[I][callback.cpp:95] callback_register(): register callback_func for powermgm success (0x400e7e20:rtcctl)
[I][callback.cpp:95] callback_register(): register callback_func for powermgm loop success (0x400e7f1c:rtcctl loop)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c595  PS      : 0x00060530  A0      : 0x800e9d84  A1      : 0x3ffd1f80  
A2      : 0x3f8284c5  A3      : 0x00000000  A4      : 0x0000003f  A5      : 0x00000001  
A6      : 0x3f4fab49  A7      : 0x3f4fa7c1  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x0000000a  
A14     : 0x00000001  A15     : 0xffffff80  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffff3  

Backtrace: 0x4000c595:0x3ffd1f80 0x400e9d81:0x3ffd1fa0 0x400e9fcd:0x3ffd2090 0x400e7a55:0x3ffd20e0 0x400ea719:0x3ffd2100 0x4013c1db:0x3ffd2130 0x400934f9:0x3ffd2150
  #0  0x4000c595:0x3ffd1f80 in ?? ??:0
  #1  0x400e9d81:0x3ffd1fa0 in wifictl_load_config() at .pio/libdeps/ttgo-t-watch/ArduinoJson/src/ArduinoJson/Json/PrettyJsonSerializer.hpp:63 (discriminator 2)
  #2  0x400e9fcd:0x3ffd2090 in wifictl_setup() at .pio/libdeps/ttgo-t-watch/ArduinoJson/src/ArduinoJson/Json/PrettyJsonSerializer.hpp:63 (discriminator 2)
  #3  0x400e7a55:0x3ffd20e0 in powermgm_setup() at src/hardware/powermgm.cpp:61
  #4  0x400ea719:0x3ffd2100 in setup() at src/my-ttgo-watch.ino:92
  #5  0x4013c1db:0x3ffd2130 in loopTask(void*) at /home/lee/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:14
  #6  0x400934f9:0x3ffd2150 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
[... 29 lines cut ...]
  #1  0x400e9d81:0x3ffd1fa0 in wifictl_load_config() at .pio/libdeps/ttgo-t-watch/ArduinoJson/src/ArduinoJson/Json/PrettyJsonSerializer.hpp:63 (discriminator 2)
  #2  0x400e9fcd:0x3ffd2090 in wifictl_setup() at .pio/libdeps/ttgo-t-watch/ArduinoJson/src/ArduinoJson/Json/PrettyJsonSerializer.hpp:63 (discriminator 2)
  #3  0x400e7a55:0x3ffd20e0 in powermgm_setup() at src/hardware/powermgm.cpp:61
  #4  0x400ea719:0x3ffd2100 in setup() at src/my-ttgo-watch.ino:92
  #5  0x4013c1db:0x3ffd2130 in loopTask(void*) at /home/lee/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:14
  #6  0x400934f9:0x3ffd2150 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
starting t-watch V1, version: 2020092401 core: 1
Configure watchdog to 30s: 0
[I][callback.cpp:44] callback_init(): init callback_t structure success for: powermgm
```

sharandac, 2020-09-25:
There are still 19 entries missing in the Json which do not exist. This makes the config loader crash. Is fixed in 2020092501.

`  #1  0x400e9d81:0x3ffd1fa0 in wifictl_load_config() at .pio/libdeps/ttgo-t-watch/ArduinoJson/src/ArduinoJson/Json/PrettyJsonSerializer.hpp:63 (discriminator 2)`

relevant code in wifictl.cpp

```
for ( int i = 0 ; i < NETWORKLIST_ENTRYS ; i++ ) {
        strlcpy( wifictl_networklist[ i ].ssid    , doc["networklist"][ i ]["ssid"], sizeof( wifictl_networklist[ i ].ssid ) );
        strlcpy( wifictl_networklist[ i ].password, doc["networklist"][ i ]["psk"], sizeof( wifictl_networklist[ i ].password ) );
}
```
fixed code that checks if the entry exists.
```
for ( int i = 0 ; i < NETWORKLIST_ENTRYS ; i++ ) {
       if ( doc["networklist"][ i ]["ssid"] && doc["networklist"][ i ]["psk"] ) {
             strlcpy( wifictl_networklist[ i ].ssid    , doc["networklist"][ i ]["ssid"], sizeof( wifictl_networklist[ i ].ssid ) );
             strlcpy( wifictl_networklist[ i ].password, doc["networklist"][ i ]["psk"], sizeof( wifictl_networklist[ i ].password ) );
       }
}
```

yourchanges, 2020-09-25:
or a workaround json:
```
{
  "autoon": true,
  "webserver": true,
  "networklist": [
    {
      "ssid": "xxxxxxx",
      "psk": "xxxx"
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    },
    {
      "ssid": "",
      "psk": ""
    }
  ]
}
```

yourchanges, 2020-09-25:
And another question: "Can we have a switch "always on" in wlan settings?", so I can make mqtt listener to watch some events from mqtt server indoor.

geekahertz, 2020-09-25:
> or a workaround json:
>
> ```
> {
>   "autoon": true,
>   "webserver": true,
>   "networklist": [
>     {
>       "ssid": "xxxxxxx",
>       "psk": "xxxx"
>     },
>     {
>       "ssid": "",
>       "psk": ""
>     }, [...]

My bad, you are right.  I think I also modified this file
My-TTGO-Watch\src\hardware\wifictl.h (line 29)
``
#define NETWORKLIST_ENTRYS          1
``
It is more of a hack (I also got timesync.json and blectl.json... oops).  I think rnisthal's post of setup thru  http://www.my-ttgo-watch.co.uk/ is the intended way.  I just didn't know that site existed before.

yourchanges, 2020-09-27:
I didn't send the config via bluetooth  gatt successfully yet. any examples or tools?

sharandac, 2020-09-27:
each message begin with a \x0a and ends with a \x10. \x0a means DataLinkEscape and \x10 means LineFeed. see hardware/blectl.cpp line 177-212

commit 09bd2552 (2020-09-24): fix #120, update icon and bluetooth msg
commit 4fe8df51 (2020-09-25): fix #120 crash while load missing network entrys
commit f1398d51 (2020-09-25): fix #120 crash while load missing network entrys
closed 2020-09-29 by sharandac