sharandac/My-TTGO-Watch #122: A lot of errors

Kabron287, 2020-09-25:
src\hardware\rtcctl.cpp:64:71: error: 'esp_sleep_enable_gpio_wakeup' was not declared in this scope
                                         esp_sleep_enable_gpio_wakeup ();
                                                                                             ^
src\hardware\powermgm.cpp:132:35: error: 'esp_light_sleep_start' was not declared in this scope
             esp_light_sleep_start();
                                              ^
src\hardware\pmu.cpp:203:35: error: 'esp_sleep_enable_gpio_wakeup' was not declared in this scope
     esp_sleep_enable_gpio_wakeup ();
                                                        ^
.pio/libdeps/ttgo-t-watch/ESP8266Audio/src/AudioFileSourceFS.h:32:26: error: expected ')' before '&' token
     AudioFileSourceFS(FS &fs) { filesystem = &fs; }
                                        ^
.pio/libdeps/ttgo-t-watch/ESP8266Audio/src/AudioFileSourceFS.h:33:26: error: expected ')' before '&' token
     AudioFileSourceFS(FS &fs, const char *filename);
                                        ^
.pio/libdeps/ttgo-t-watch/ESP8266Audio/src/AudioFileSourceFS.h:45:5: error: 'FS' does not name a type
     FS *filesystem;
     ^
.pio/libdeps/ttgo-t-watch/ESP8266Audio/src/AudioFileSourceFS.h:46:5: error: 'File' does not name a type
     File f;
     ^
.pio/libdeps/ttgo-t-watch/ESP8266Audio/src/AudioFileSourceFS.h:42:47: error: 'f' was not declared in this scope
     virtual uint32_t getPos() override { if (!f) return 0; else return f.position(); };
                                                                 ^
.pio/libdeps/ttgo-t-watch/ESP8266Audio/src/AudioFileSourceSPIFFS.h:33:55: error: no matching function for call to 'AudioFileSourceFS::AudioFileSourceFS(fs::SPIFFSFS&)'
     AudioFileSourceSPIFFS() : AudioFileSourceFS(SPIFFS) { };
                                                                                       ^
.pio/libdeps/ttgo-t-watch/ESP8266Audio/src/AudioFileSourceSPIFFS.h:34:85: error: no matching function for call to 'AudioFileSourceFS::AudioFileSourceFS(fs::SPIFFSFS&, const char*&)'
     AudioFileSourceSPIFFS(const char *filename) : AudioFileSourceFS(SPIFFS, filename) {};
                                                                                                                                       ^

sharandac, 2020-09-25:
have you read the readme.md on the start page and followed the instructions?

Kabron287, 2020-09-26:
What exactly I did wrong?
I only could upload binary without build.
But I want to build it.

sharandac, 2020-09-26:
Do you use PlatformIO or the ArduinoIDE?

Kabron287, 2020-09-26:
Surely, I use platformio!
It could be seen by **.pio** prefix in error messages.

sharandac, 2020-09-26:
I just wanted to make sure. I'm a bit surprised by the error message, it only appears if the esp32 framework is not installed. but that should happen automatically when the project is opened.

please try to insert the following line at the top of bma.cpp

`#include "esp_pm.h"`

Kabron287, 2020-09-26:
Did not help. And a lot of errors conserning: UBaseType_t, TickType_t, BaseType_t, expected ')' before numeric constant.

sharandac, 2020-09-26:
i just installed vscode and platformio under windows and cloned the project. after a few minutes i had a successful build without any errors. strange.

Kabron287, 2020-09-26:
Interesting, what tools you were used before?

sharandac, 2020-09-26:
the same tool under windows and linux. for this case i used a fresh windows 10 installation and had only installed vscode with platformio. normally i used linux to build, this is much faster.

Kabron287, 2020-09-26:
I have no ideas...

sharandac, 2020-09-26:
do you have the possibility to test it on another machine?

robouden, 2020-09-26:
I run VSCodium on two computers one IBM and other MacBook Pro. Both no issues at all. VSCodium 1.49.2  and PlatformIO (3.3.0).

Kabron287, 2020-09-26:
Strange, on my office PC I firstly got 3 errors with **esp_sleep_enable_gpio_wakeup**
with **#include "esp_pm.h"** I got huge amount of types errors.
So, I deleted #include and now got only one **esp_sleep_enable_gpio_wakeup** error in bma.cpp

Kabron287, 2020-09-26:
Very strange. Each compilation I got different error results.
One time only esp_sleep, another with types...

sharandac, 2020-09-26:
Strange ... i am not sure if vscodium supports platformio. i have problems to install platformio in vscodium. searching the internet i found out that there are some problems.

Kabron287, 2020-09-26:
Sorry, I confused TeamViewer window with local. On office PC was only one error with ESP8266Spiram. After installig the Lib compilaton OK.
I suspect somethig wrong with libs on my local PC, because I see stange link to Seeed libraries in compilaton results.

sharandac, 2020-09-26:
Okay. I would still like to know what the ESP8266Spiram is all about. I have never seen this message with VSCode. Could it be related to VSCodium?

Kabron287, 2020-09-26:
Bingo! It works on my home PC.
Obviously, Seeed libs are very poisonous ones.

Kabron287, 2020-09-26:
> Okay. I would still like to know what the ESP8266Spiram is all about. I have never seen this message with VSCode. Could it be related to VSCodium?

Only now hear about VSCodium. Interesting to try.

sharandac, 2020-09-26:
this is the first time I have heard of VSCodium. The first impression was not the best :-)

robouden, 2020-09-26:
Not much different then VSCode.. But open-source and community-supported. Prefer that takes nearly all the extensions a VSCOde (except you have to hack the market place to get PlatformIO installed and to have an error being suppressed, That seems still MS way to keep control)

robouden, 2020-09-26:
The hack is easy.. edit product.json and replace the store with 

```
"extensionsGallery": {
    "serviceUrl": "https://marketplace.visualstudio.com/_apis/public/gallery",
    "itemUrl": "https://marketplace.visualstudio.com/items"
}
```

closed 2020-09-30 by sharandac