sharandac/My-TTGO-Watch #124: Daylight Savings switch should apply straight away

datacute, 2020-09-27:
New Zealand changed to daylight savings time this morning.
I'm aware that setting an offset and a switch is easier than dealing with timezone rules, however...
I wasn't able to get the watch to display the correct time until I got home, and could connect to wifi with the time set to sync when connected.
Changing the daylight savings switch on or off should take effect straight away.

On a related note but different issue: there are timezones with values for offsets from UTC that are not -12 to +12 whole hours, e.g. India using +5:30, Tonga using +13:00
I've used https://github.com/JChristensen/Timezone for an esp8266 controlled clock before - is there interest in incorporating that library here?

datacute, 2020-09-28:
Scratch that Timezone library idea... using the ESP's configTzTime looks like a better idea.
Relevant Timezone data is available in csv or json from https://github.com/nayarsystems/posix_tz_db

robouden, 2020-09-28:
Agree, much easier.

datacute, 2020-09-28:
I'm half way through implementing a change to:
1: embed timezones rules json (done)
2: parse timezones json to build drop down list of time zone rules instead of hours offsets from utc (done)
3: remove daylight savings checkbox (done)
4: set timezones from selected rule (to do, proof of concept works)
5: test that daylight savings switch happens automatically (to do)
6: make timezone selection apply straight away (to do)
7: upgrade old config settings to etc/gmt-x (to do)

(5 and 6 solve this issue)

I've never used LVGL, so expect someone else will clean up any annoying alignment/behavior. It'd be nice to be able to select region and rule in two drop-down boxes instead of one massive one.

sharandac, 2020-09-29:
Great! Please insert your code into the experimental branch. We will do the cleanup there.

datacute, 2020-09-30:
Pull request #126 created.
I realised I had to convert the RTC store of time to UTC else strange things happen. I think that solves the 1 hour off problems recently mentioned in telegram too).
I tested with the daylight savings time starting, ending, and an hour after ending (when 3am is reached for the second time that morning).
Since this basically needs a time sync to sort it out, I've not implemented auto-upgrade of the config.

datacute, 2020-09-30:
I'd include a screenshot or two - but that appears broken for me at the moment.

datacute, 2020-10-02:
It takes a LOT of scrolling to get down to Pacific region.
Does anyone want to rewrite the selection so that it is two selection boxes? The first selecting from the string "Africa\nAmerica\nAntarctica\nArtic\nAisa\nAtlantic\nAustralia\nEtc\nEurope\nIndian\nPacific"
[image]
[image]

sharandac, 2020-10-02:
This is really a LOT to scroll! I'm putting that on my todo list for next week.

closed 2020-10-08 by sharandac