sharandac/My-TTGO-Watch #127: alarm is not raised

jakub-vesely, 2020-09-30:
Dear Dirk, 
I would like to report that alarm clock on master  does not work on my HW and was broken by commit 62100ff3df6b89421f874a6daeb8930967126199
I expect that it was because you commented out ttgo->rtc->syncToRtc() in rtcctl. Caused these line an issue for you?

b.t.w. there is more changes that i do not understand in the commit. I would like to follow your rules in your repository. Would be good to you publish some guidelines or highlight them during the integration process. 

Thank you, Jakub

jakub-vesely, 2020-09-30:
I tried to uncomment but is still does not work it is probably another from your changes.

sharandac, 2020-09-30:
Hey Jakub,

this line of code was an issue, for me. In the experiemental branch i have reduce the interaction between rtc and the esp to a minimum to resolve some problems. At least, only two interactions was the result. syncToSystem() after boot or an wakeup to sync the system time and only one syncToRtc() after an timesync via Wifi. From here we never touch syncToRtc() again. The idea was an better and stable time handling. At to same time, @datacute work on an better timezone support. And here comes the important part as an result of it, the rtc store the time in GMT0. It results in an significant better timezone handling.

After this change i have merge your code and run into non working alarm function. I have fix it with version 2020093003, see https://github.com/sharandac/My-TTGO-Watch/commit/c1c86114f10d202afc5a68e648626332c462b5a0. The important code is between line 138 and 147.

Hope I didn't scare you! And you are right, some code guidelines and hints from me during the integration process is an helpful information. My fault. And if you like, we can write directly via telegram or use jitsi.

thanks a lot for your code,
Dirk

datacute, 2020-10-01:
@sharandac you can use gmtime_r instead of localtime_r avoiding the need to change timezone twice.

datacute, 2020-10-01:
rtcctl_setup() should be after timesync_setup()

sharandac, 2020-10-01:
right, that would be a good idea.

jakub-vesely, 2020-10-01:
Hi Dirk,
I understand that the situation was not simple and there was more branches affected the same thing and you have to manage them somehow. Shall I ask for the merge to the experimental branch instead of master next time? would be better for you?  

I tried the version with the commit 83de621 but the issue is still there. The alarm_in_progress tile is not opened when it is expected. Shall I take a look at it or do you prefer to solve it yourself? 

Please, do not shy to raise an issue during the integration procedure for my pull requests.
From my point of you it is better to raise and discuss the issue instead of accept everything how is asked and then to change more than 100 lines of just accepted code. To be honest, I have to say that I'm a little bit confused now.  

Unfortunately, I do not have telegram or jitsi accounts, I'm sorry. Maybe I will create some of them later.

Thank you for your leading of this project. It is awesome!

Jakub

sharandac, 2020-10-01:
Yes, it is not an easy task to merge different codes with different code styles. For the future it is better to use always in experimental branch to merge. After that we can quietly search for errors. And it is very important to keep the experimental branch locally up-to-date.

Interesting, on my watch the alarm works perfectly. I have tested it with different time zones.
Did you set your time zone correctly after the update and re-synchronise it over wifi?
If an alarm is set this message should appear:
```
[I][rtcctl.cpp:126] set_next_alarm(): local time: 12:00 day: 1
[I][rtcctl.cpp:137] set_next_alarm(): next local alarm time: 12:08 day: 1
[I][rtcctl.cpp:139] set_next_alarm(): next GMT0 alarm time: 10:08 day 1
```

jakub-vesely, 2020-10-01:
Ok, I will add more logs and try to find out why it does not work for me.

sharandac, 2020-10-01:
I have now tested four watches with four different time zones and different alarm times. Everywhere so far it has worked as expected. Strange. I am a bit overwhelmed.

jakub-vesely, 2020-10-01:
Dear Dirk,
I tested couple of times during the day and can confirm that it works for me as well. I have never seen that issue what I had in the morning. It had to be a border issue after first setting of the timezone or something related to first time synchronization. It happens sometimes that time is not synchronized as well and the original time remains.  I was able reproduce it in the morning but I did not have required log to localize the issue.

I'm going to close this issue and will focus on this marginal case in following days and try to repeat it. 

Thank you for your effort in solving my issue.

closed 2020-10-01 by jakub-vesely